platformio/platformio-atom-ide #319: exception opening project

Spongman, 2017-04-01:
not sure how to repro this, but after launching the ide and trying to open a project from the "recent projects" list, the debug console showed this:

````
uncaught at root 
 at watchOpenProject 
 at takeEveryHelper 
 at  
 TypeError: Cannot read property 'Symbol(Symbol.iterator)' of undefined
    at setObjectMembers (C:\Users\PiersH\AppData\Local\atom\app-1.15.0\resources\electron.asar\renderer\api\remote.js:98:22)
    at loadRemoteProperties (C:\Users\PiersH\AppData\Local\atom\app-1.15.0\resources\electron.asar\renderer\api\remote.js:164:5)
    at Object.getOwnPropertyDescriptor (C:\Users\PiersH\AppData\Local\atom\app-1.15.0\resources\electron.asar\renderer\api\remote.js:179:7)
    at undefined.descriptor.get [as setJumpList] (C:\Users\PiersH\AppData\Local\atom\app-1.15.0\resources\electron.asar\renderer\api\remote.js:118:32)
    at ReopenProjectMenuManager.updateWindowsJumpList (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\src\reopen-project-menu-manager.js:111:15)
    at ReopenProjectMenuManager.update (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\src\reopen-project-menu-manager.js:79:12)
    at Function.module.exports.Emitter.simpleDispatch (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\node_modules\event-kit\lib\emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\node_modules\event-kit\lib\emitter.js:129:28)
    at HistoryManager.didChangeProjects (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\src\history-manager.js:77:20)
    at HistoryManager.addProject (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\src\history-manager.js:95:12)
    at Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\src\history-manager.js:36:20
    at Function.module.exports.Emitter.simpleDispatch (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\node_modules\event-kit\lib\emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\node_modules\event-kit\lib\emitter.js:129:28)
    at Project.module.exports.Project.addPath (Z:\PiersH\AppData\Local\atom\app-1.15.0\resources\app.asar\src\project.js:288:29)
    at file:///C:/Users/PiersH/.atom/packages/platformio-ide/lib/project/sagas.js:46:18
    at createTaskIterator (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\proc.js:158:17)
    at runForkEffect (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\proc.js:541:24)
    at runEffect (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\proc.js:432:577)
    at next (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\proc.js:313:9)
    at currCb (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\proc.js:386:7)
    at takeCb (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\proc.js:457:108)
    at Object.put (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\channel.js:87:16)
    at C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\channel.js:163:10
    at C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\channel.js:190:16
    at exec (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\scheduler.js:28:5)
    at asap (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\scheduler.js:39:5)
    at Array.<anonymous> (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\channel.js:189:27)
    at Object.emit (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\channel.js:40:13)
    at C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux-saga\lib\internal\middleware.js:81:21
    at Object.onOpen (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\redux\lib\bindActionCreators.js:7:12)
    at onClick (file:///C:/Users/PiersH/.atom/packages/platformio-ide/lib/project/components/projects-block.js:91:36)
    at Object.invokeGuardedCallback (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactErrorUtils.js:26:5)
    at executeDispatch (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\EventPluginUtils.js:85:21)
    at Object.executeDispatchesInOrder (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\EventPluginUtils.js:108:5)
    at executeDispatchesAndRelease (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\EventPluginHub.js:43:22)
    at executeDispatchesAndReleaseTopLevel (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\EventPluginHub.js:54:10)
    at Array.forEach (native)
    at forEachAccumulated (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\forEachAccumulated.js:24:9)
    at Object.processEventQueue (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\EventPluginHub.js:257:7)
    at runEventQueueInBatch (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactEventEmitterMixin.js:17:18)
    at Object.handleTopLevel [as _handleTopLevel] (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactEventEmitterMixin.js:28:5)
    at handleTopLevelImpl (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactEventListener.js:72:24)
    at ReactDefaultBatchingStrategyTransaction.perform (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\Transaction.js:140:20)
    at Object.batchedUpdates (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactDefaultBatchingStrategy.js:62:26)
    at Object.batchedUpdates (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactUpdates.js:97:27)
    at dispatchEvent (C:\Users\PiersH\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactEventListener.js:147:20)
````

after that, i wasn't able to open a project folder any way...

ivankravets, 2017-04-01:
> after that, i wasn't able to open a project folder any way...

even after close/open Home?

ivankravets, 2017-05-06:
Should be fixed in the latest PIO IDE 2.0 Beta 3

closed 2017-05-06 by ivankravets