platformio/platformio-atom-ide #320: Failed to run a command: build

dsjstc, 2017-04-01:
New user running on Ubuntu 16.10, IDE  2.0.0-beta.2.3, Core 3.4.0a3.  Getting the "failed to run a command: build" error.

**What I'm doing:** I followed the quickstart recipe, and tried with a new project, and importing an existing Arduino IDE project.  Deleted my .atom and .platformio directories and reinstalled everything.  No joy.

**What I've tried:** 
- Verified I don't have a separate version of PIO.
- Downgraded to ide 1.7.3 (didn't work for me at all).
- Checked the FAQ, this error is not listed.
- Checked build package is enabled, per [bug #79](https://github.com/platformio/platformio-atom-ide/issues/79)
- Tried with platformio-build package instead.
- Verified that the same install sequence works as expected on Windows 10 x64.

ivankravets, 2017-04-01:
Could you make clean uninstall of PIO IDE + Atom? Then, install the latest Atom and stable PIO IDE 2.0.0-beta.2.

http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

dsjstc, 2017-04-01:
No joy with three different versions of atom.   I'm not familiar with Atom, is there a process log somewhere?

NB that some elements of the clean install instructions are absent on my computer:

- .config/Atom exists, but not .config/Atom-Shell
- I am using packaged versions of Atom, so nothing is in /usr/local/...

I attempted with the following versions of Atom, with full purge (including apt-get purge) before each attempt.

- initial attempt: atom-amd64 1.15.0 from webupd8 ppa
- second attempt: 1.15.0 direct download from front page of atom.io
- third attempt: 1.16.0-beta0 from beta page of atom.io.

ivankravets, 2017-04-02:
Could you provide an output of `apm list` command?

dsjstc, 2017-04-02:
bash: .bashrc: No such file or directory
at@punk:~/pblink$

[36mBuilt-in Atom Packages[39m (89)
├── atom-dark-syntax@0.28.0
├── atom-dark-ui@0.53.0
├── atom-light-syntax@0.29.0
├── atom-light-ui@0.46.0
├── base16-tomorrow-dark-theme@1.5.0
├── base16-tomorrow-light-theme@1.5.0
├── one-dark-ui@1.9.1
├── one-light-ui@1.9.1
├── one-dark-syntax@1.7.1
├── one-light-syntax@1.7.1
├── solarized-dark-syntax@1.1.2
├── solarized-light-syntax@1.1.2
├── about@1.7.2
├── archive-view@0.62.2
├── autocomplete-atom-api@0.10.0
├── autocomplete-css@0.15.0
├── autocomplete-html@0.7.2
├── autocomplete-plus@2.34.2
├── autocomplete-snippets@1.11.0
├── autoflow@0.29.0
├── autosave@0.24.0
├── background-tips@0.26.1
├── bookmarks@0.44.1
├── bracket-matcher@0.85.2
├── command-palette@0.40.1
├── deprecation-cop@0.56.2
├── dev-live-reload@0.47.0
├── encoding-selector@0.23.1
├── exception-reporting@0.41.0
├── find-and-replace@0.206.3
├── fuzzy-finder@1.4.1
├── git-diff@1.3.1
├── go-to-line@0.32.0
├── grammar-selector@0.49.1
├── image-view@0.60.0
├── incompatible-packages@0.26.1
├── keybinding-resolver@0.36.1
├── line-ending-selector@0.6.1
├── link@0.31.2
├── markdown-preview@0.159.6
├── metrics@1.1.3
├── notifications@0.66.2
├── open-on-github@1.2.1
├── package-generator@1.1.0
├── settings-view@0.247.0
├── snippets@1.0.5
├── spell-check@0.70.2
├── status-bar@1.8.1
├── styleguide@0.49.2
├── symbols-view@0.114.0
├── tabs@0.104.1
├── timecop@0.34.0
├── tree-view@0.214.1
├── update-package-dependencies@0.10.0
├── welcome@0.36.0
├── whitespace@0.36.2
├── wrap-guide@0.39.0
├── language-c@0.56.0
├── language-clojure@0.22.2
├── language-coffee-script@0.48.5
├── language-csharp@0.14.2
├── language-css@0.42.0
├── language-gfm@0.88.1
├── language-git@0.19.0
├── language-go@0.43.1
├── language-html@0.47.2
├── language-hyperlink@0.16.1
├── language-java@0.26.0
├── language-javascript@0.126.1
├── language-json@0.18.3
├── language-less@0.30.1
├── language-make@0.22.3
├── language-mustache@0.13.1
├── language-objective-c@0.15.1
├── language-perl@0.37.0
├── language-php@0.37.4
├── language-property-list@0.9.0
├── language-python@0.45.2
├── language-ruby@0.70.5
├── language-ruby-on-rails@0.25.2
├── language-sass@0.57.1
├── language-shellscript@0.25.0
├── language-source@0.9.0
├── language-sql@0.25.3
├── language-text@0.7.1
├── language-todo@0.29.1
├── language-toml@0.18.1
├── language-xml@0.34.16
└── language-yaml@0.28.0

[36mCommunity Packages (13)[39m /home/at/.atom/packages
├── autocomplete-clang@0.10.0
├── build@0.67.0
├── busy@0.7.0
├── file-icons@2.1.2
├── intentions@1.1.2
├── language-ini@1.19.0
├── linter@2.1.2
├── linter-gcc@0.7.1
├── linter-ui-default@1.2.2
├── minimap@4.27.0
├── platformio-ide@2.0.0-beta.2
├── platformio-ide-terminal@2.5.0
└── tool-bar@1.1.0

dsjstc, 2017-04-02:
I've isolated this and worked around it:  My dev directory is a symlink.  I had opened the project through that symlink.  When opened with the full absolute filepath with no symlinks, it works correctly.

Suggest you retain this as a bug and add some liberal use of $(readlink -f $1) to the code.

theopensourcerer, 2017-04-08:
@dsjstc I could kiss you! This was driving me nuts. I uninstalled and re-installed platformio, tried all dorts of things but kept getting the "Failed to run a command build" error. Even though Build was installed and enabled.

I just read your message about symlinks :-) My dev directory structure is on a separate disk and I symlink it to a local directory. Guess what. I closed the project, and opened it via the full path and it works! I didn't have this issue before the recent IDE 2 upgrade

Thanks a million for that tip.

PR #427 (2017-05-01): Avoid unnecessary realpath() calls
According to atom/atom#9879, TextEditor.getPath() does not return a realpath
since Atom 1.14. We no longer need to normalize paths by calling realpath() in
order to determine the active project, and we can use the actual project paths
and text editor paths instead.

Resolve #136, Resolve #320.
closed 2017-05-01 by ivankravets