platformio/platformio-atom-ide #323: Install missed packages: linter-ui-default,intentions

Yojamac, 2017-04-02:
Error: events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: ENOENT: no such file or directory, open '/Users/jayo2/.atom/packages/linter-ui-default/lib/panel/component.js'
    at Error (native)

    at /Users/jayo2/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:86:27
    at onExit (/Users/jayo2/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (/Users/jayo2/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:322:11)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:343:11
    at Socket.<anonymous> (/Applications/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:204:9)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

closed 2017-04-02 by Yojamac

ivankravets, 2017-04-02:
How did you resolve that?

Yojamac, 2017-04-05:
Complete un-install of Atom as given in the PlatformIO Docs and then re-installed.
http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide