platformio/platformio-atom-ide #333: Serial Monitor: pio: command not found

kalsan, 2017-04-04:
```
Atom: 1.15.0 x64 Linux
Platformio: IDE 2.0.0-beta.2 · Core 3.4.0a4
```

Steps to reproduce:
-> Press Serial Monitor
-> Select any port and baud rate
-> The following error occurs
```
pio device monitor --port /dev/ttyS0 --filter colorize --eol LF --echo
[me@laptop Directory] pio device monitor --port /dev/ttyS0 --filter colorize --eol LF --echo
bash: pio: command not found
```

(Build and Upload steps work fine though).

tehnorm, 2017-04-06:
I am also seeing the same issue.

```
Atom: 1.15.0 x64 OSX
Platformio: IDE 2.0.0-beta.2.3 · Core 3.4.0a4
```

tehnorm, 2017-04-06:
There seems to be a work around: https://github.com/platformio/platformio-atom-ide/issues/47#issuecomment-192302365

The path is different for the beta versions: `Menu: PlatformIO > Settings > PlatformIO IDE Terminal > Auto Run Command and specify it to export PATH=export PATH=~/.platformio/penv/bin:$PATH`

kalsan, 2017-04-06:
@tehnorm thanks for sharing this workaround, it helps.

ivankravets, 2017-04-06:
The problem is linked with customized `.bash_profile`. Please check that you don't override `PATH` variable. You need to keep previous value of `PATH`. For example, use 
```
export PATH="/cusotm/paths:${PATH}"
```
instead of 

```
export PATH="/cusotm/paths"
```

kalsan, 2017-04-09:
Hmm, I don't actually override the PATH variable in bashrc or bash_profile.

In bashrc I simply have: `export PATH=$PATH:'/home/sandro/.gem/ruby/2.3.0/bin'`

And this is my acual path variable:
```
echo $PATH
/home/user/.gem/ruby/2.4.0/bin:/home/user/.gem/ruby/2.3.0/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/share/java/basex/bin:/usr/lib/jvm/default/bin:/opt/opencascade/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/user/.gem/ruby/2.3.0/bin
```

ivankravets, 2017-06-01:
Documented in http://docs.platformio.org/en/latest/faq.html#platformio-is-not-recognized-as-an-internal-or-external-command

kalsan, 2017-06-02:
Why closing? The problem is not resolved, as stated in my comment on April 9th

ivankravets, 2017-06-02:
@kalsan what is your shell?

kalsan, 2017-06-03:
Thanks @ivankravets 

I'm using bash on login. Sometimes I use fish, but it's unrelated to atom / platformio activities.

ivankravets, 2017-06-03:
@kalsan what is your default shell? Could you check Fish initial configuration? Somewhere you override global $PATH.

kalsan, 2017-06-04:
- My default shell is bash
- config.fish only contains a source with aliases
- fish handles paths specially: there is a $fish_user_paths which gets automatically prepended to the $PATH variable in fish. Digging through all configs, I can't find a single override.

When / how does platformio extend the PATH variable? Perhaps I can try to debug it this way.

ivankravets, 2017-06-05:
@kalsan could you try [this](http://georgik.rocks/workaround-failed-to-activate-the-platformio-ide-package/) solution?

> When / how does platformio extend the PATH variable? Perhaps I can try to debug it this way.

PIO IDE extends PATH with a path to PIO Core before starting a terminal instance. If you override PATH in your shell scripts, you will not see PIO Core here. You can try manually to make a few `echo $PATH` in different places.

kalsan, 2017-06-05:
> @kalsan could you try this solution?

```
My shell is already bash:
user@machine ~> echo $SHELL
/bin/bash
user@machine ~> sudo chsh -s /bin/bash user
[sudo] Passwort für user: 
Shell für user ändern.
chsh: Shell nicht geändert.
```
(Translation: "chsh: chell not changed")

I just installed the latest PlatformIO update. It didn't help.

This is what attempting to open a serial monitor looks like without the workaround linked by @tehnorm :
```
pio device monitor --port /dev/ttyS1 --filter colorize --eol LF --echo
[user@machine bitthief]$ pio device monitor --port /dev/ttyS1 --filter colorize --eol LF --echo
bash: pio: command not found
[user@machine bitthief]$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/share/java/basex/bin:/usr/lib/jvm/default/bin:/opt/opencascade/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr
/bin/core_perl:/home/user/.gem/ruby/2.3.0/bin
[user@machine bitthief]$
```

This is what happens if I apply the workaround:

```
export PATH=export PATH=~/.platformio/penv/bin:$PATH
pio device monitor --port /dev/ttyS1 --filter colorize --eol LF --echo
[user@machine bitthief]$ export PATH=export PATH=~/.platformio/penv/bin:$PATH
[user@machine bitthief]$ pio device monitor --port /dev/ttyS1 --filter colorize --eol LF --echo
--- Miniterm on /dev/ttyS1  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
 
--- exit ---
[user@machine bitthief]$ echo $PATH
/home/user/.platformio/penv/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/share/java/basex/bin:/usr/lib/jvm/default/bin:/opt/opencascade/bin:/usr/bin/s
ite_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/user/.gem/ruby/2.3.0/bin
[user@machine bitthief]$ ^C
[user@machine bitthief]$
```

ivankravets, 2017-06-06:
> (Translation: "chsh: shell not changed")

You said that you have a fish shell, try to change to it and back to bash:

* http://georgik.rocks/workaround-failed-to-activate-the-platformio-ide-package/

------

Could you place this `export PATH=~/.platformio/penv/bin:$PATH` in your bash profile?

kalsan, 2017-06-12:
@ivankravets here's a log of the experiments I did following your suggestions:

**1. Changing shell**
- Changing shell to `/usr/bin/fish`
    - pio works without the export workaround
- Changing shell back to '/bin/bash'
    - previous behaviour restored: 'pio: command not found'

**2. Adding the penv path to '.bashrc'
- This is a valid workaround, works without the export workaround in the pio terminal settings

The second option would be an acceptable workaround for me, since it survives pio updates (unlike the previous export workaround). If you are satisfied with this solution as well, the issue can be closed in my opinion. However if you'd like to invstigate further to get more insights an maybe find a potential bug, I'll be happy to assist.

kalsan, 2017-10-03:
Issue still exists in 2.0.0-rc2

Thijxx, 2017-11-17:
issue still exists in 2.0.0-rc4

VanKurt, 2017-11-30:
I have the same problem (just installed PlatformIO on macOS). When opening the serial monitor I get:

> pio device monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
> Boriss-Air:RobotArm borisbrock$ pio device monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
> bash: pio: command not found

VanKurt, 2017-12-11:
Any updates? I still can't work on my project because of this :-(

Thijxx, 2017-12-11:
Does the use of the `export` command not work for you?

ivankravets, 2017-12-11:
@VanKurt do you use a custom shell?

ivankravets, 2017-12-12:
@VanKurt 
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

VanKurt, 2017-12-12:
Hi,
here's what I get:

> Boriss-Air:~ boris$ echo $PATH
> /Library/Frameworks/Python.framework/Versions/3.4/bin:/opt/local/bin:/opt/local/sbin:/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/tools:/platform-tools
> Boriss-Air:~ boris$ echo $PYTHONPATH
> 
> Boriss-Air:~ boris$ python --version
> Python 2.7.10
> Boriss-Air:~ boris$ pip --version
> -bash: pip: command not found
> Boriss-Air:~ boris$ pip list
> -bash: pip: command not found

herm, 2018-01-19:
I also had this problem on a fresh install of Manjaro Linux (based on Arch). I did no customization wrt PATH. Compiling and programming works fine but serial monitor doesn't find pio. For me the soultion was to set Settings->Platform IO Terminal->Login shell to false. Probably somewhere in the systemwide config the path is overwritten for login shells.

Sufomba, 2018-02-02:
Same problem with MX Linux, I did what 'herm' did and now it is working fine.

dvera2004, 2018-02-12:
Hi,

For fix it, please make a symlink of pio to /bin

sudo ln -s /home/daniel/.platformio/penv/bin/pio /bin/pio

Works perfect.

Regards

itsjimbo, 2018-08-13:
Setup bashrc path
```
echo "export PATH=~/.platformio/penv/bin:$PATH" >> ~/.bashrc
source .bashrc
```
Now test
```
cd my-platformio-project
pio run -t uploadfs
```

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets