platformio/platformio-atom-ide #335: Uncaught Error: Cannot find module 'fs-plus'

deladriere, 2017-04-04:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.15.0 x64
**Electron**: 1.3.13
**OS**: Mac OS X 10.12.4
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.2.3

### Stack Trace

Uncaught Error: Cannot find module 'fs-plus'

```
At module.js:457

Error: Cannot find module 'fs-plus'
    at Module._resolveFilename (module.js:455:15)
    at Module._resolveFilename (/Applications/Atom.app/Contents/Resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._resolveFilename (/app.asar/src/module-cache.js:383:52)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/platformio-ide/lib/config.js:17:15)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/platformio-ide/lib/main.js:25:15)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/app.asar/src/package.js:796:27)
    at Package.module.exports.Package.activateConfig (/app.asar/src/package.js:266:12)
    at PackageManager.module.exports.PackageManager.packageHasSettings (/app.asar/node_modules/settings-view/lib/package-manager.js:53:14)
    at PackageCard.module.exports.PackageCard.hasSettings (/app.asar/node_modules/settings-view/lib/package-card.js:728:34)
    at PackageCard.module.exports.PackageCard.updateSettingsState (/app.asar/node_modules/settings-view/lib/package-card.js:402:16)
    at PackageCard.module.exports.PackageCard.updateInterfaceState (/app.asar/node_modules/settings-view/lib/package-card.js:395:12)
    at PackageCard.module.exports.PackageCard.initialize (/app.asar/node_modules/settings-view/lib/package-card.js:243:19)
    at /app.asar/node_modules/space-pen/lib/space-pen.js:184:25)
    at /app.asar/node_modules/settings-view/lib/package-card.js:20:48)
    at PackageDetailView.module.exports.PackageDetailView.completeInitialzation (/app.asar/node_modules/settings-view/lib/package-detail-view.js:152:28)
    at PackageDetailView.module.exports.PackageDetailView.loadPackage (/app.asar/node_modules/settings-view/lib/package-detail-view.js:171:21)
    at PackageDetailView.module.exports.PackageDetailView.initialize (/app.asar/node_modules/settings-view/lib/package-detail-view.js:146:12)
    at /app.asar/node_modules/space-pen/lib/space-pen.js:184:25)
    at /app.asar/node_modules/atom-space-pen-views/lib/scroll-view.js:12:47)
    at /app.asar/node_modules/settings-view/lib/package-detail-view.js:38:54)
    at /app.asar/node_modules/settings-view/lib/settings-view.js:269:22
    at SettingsView.module.exports.SettingsView.getOrCreatePanel (/app.asar/node_modules/settings-view/lib/settings-view.js:274:19)
    at SettingsView.module.exports.SettingsView.showPanel (/app.asar/node_modules/settings-view/lib/settings-view.js:320:24)
    at SettingsView.module.exports.SettingsView.showDeferredPanel (/app.asar/node_modules/settings-view/lib/settings-view.js:315:19)
    at SettingsView.module.exports.SettingsView.initializePanels (/app.asar/node_modules/settings-view/lib/settings-view.js:162:12)
    at /app.asar/node_modules/settings-view/lib/settings-view.js:86:24
    at _combinedTickCallback (internal/process/next_tick.js:67:7)
    at process._tickCallback (internal/process/next_tick.js:98:9)

```

### Commands

```
     -0:05.5.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-sunburst.theme-one-light-ui)
```

### Non-Core Packages

```
atom-beautify 0.29.17 
autocomplete-clang 0.10.1 
build 0.67.0 
busy 0.7.0 
busy-signal 1.3.0 
dbg 1.4.1 
dbg-gdb 1.6.2 
file-icons 2.1.2 
foldingtext-for-atom 0.9.2 
intentions 1.1.2 
language-ini 1.19.0 
language-openscad 0.4.3 
linter 2.1.2 
linter-gcc 0.7.1 
linter-ui-default 1.2.2 
minimap 4.27.0 
minimap-titles 1.11.0 
output-panel 0.1.4 
platformio-ide 2.0.0-beta.2.3 
platformio-ide-terminal 2.5.0 
sunburst 0.3.0 
tool-bar 1.1.0 
```

ivankravets, 2017-04-30:
Could you reproduce this issue after restart?

ivankravets, 2017-05-06:
Should be fixed in the latest PIO IDE 2.0 Beta 3

closed 2017-05-06 by ivankravets

sabatmonk, 2017-05-20:
Im having this issue with beta 3.3
Here is the error paste
[Enter steps to reproduce:]

1. open atom
2. ...

**Atom**: 1.17.0 x64
**Electron**: 1.3.15
**OS**: Ubuntu 16.04.2
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.3.3

### Stack Trace

Failed to activate the platformio-ide package

```
At Cannot find module 'fs-plus'

Error: Cannot find module 'fs-plus'
    at Module._resolveFilename (module.js:455:15)
    at Module._resolveFilename (/usr/share/atom/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (/usr/share/atom/resources/app/src/module-cache.js:354:1)
    at Module.require (usr/share/atom/resources/app/static/index.js:40:43)
    at require (/usr/share/atom/resources/app/src/native-compile-cache.js:66:1)
    at /packages/platform-atom-ide/lib/config.js:17:15)
    at Module.get_Module._compile (/usr/share/atom/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (/usr/share/atom/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (usr/share/atom/resources/app/static/index.js:47:45)
    at require (/usr/share/atom/resources/app/src/native-compile-cache.js:66:1)
    at /packages/platform-atom-ide/lib/main.js:25:15)
    at Module.get_Module._compile (/usr/share/atom/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (/usr/share/atom/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (usr/share/atom/resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/usr/share/atom/resources/app/static/<embedded>:73:1)
    at Package.module.exports.Package.requireMainModule (/usr/share/atom/resources/app/src/package.js:887:1)
    at Package.module.exports.Package.activateNow (/usr/share/atom/resources/app/src/package.js:242:1)
    at /usr/share/atom/resources/app/src/package.js:1014:1
    at Function.module.exports.Emitter.simpleDispatch (/usr/share/atom/resources/app/node_modules/event-kit/lib/emitter.js:25:1)
    at Emitter.module.exports.Emitter.emit (/usr/share/atom/resources/app/node_modules/event-kit/lib/emitter.js:129:1)
    at PackageManager.module.exports.PackageManager.triggerDeferredActivationHooks (/usr/share/atom/resources/app/src/package-manager.js:673:1)
    at /usr/share/atom/resources/app/src/package-manager.js:604:1
```

### Commands

```
  3x -1:44.6.0 core:backspace (input.hidden-input)
     -1:43 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
platformio-ide 2.0.0-beta.3.3 
platformio-ide-terminal 2.5.0 
```

ivankravets, 2017-05-20:
@sabatmonk Please try uninstall/install `platformio-ide` package. It looks that not all packages were installed by Atom installer.

sabatmonk, 2017-05-21:
i installed manually for proxy reasons. I manage to set-up my proxy and reinstalled with atom, it solved it. Thx