platformio/platformio-atom-ide #337: InstallationManager

akkord64, 2017-04-06:
Virtualenv Download: Traceback (most recent call last):
  File "C:/Users/Chris/.platformio/penv/lib/python2.7/site.py", line 703, in <module>
    main()
  File "C:/Users/Chris/.platformio/penv/lib/python2.7/site.py", line 692, in main
    aliasmbcs()
  File "C:/Users/Chris/.platformio/penv/lib/python2.7/site.py", line 515, in aliasmbcs
    import locale, codecs
  File "C:\Users\Chris\.platformio\penv\lib\python2.7/locale.py", line 17, in <module>
    import functools
ImportError: No module named functools

ivankravets, 2017-04-06:
Could you run in system terminal `python --version`?

akkord64, 2017-04-14:
C:\Users\Chris>python --version
Python 3.6.1

C:\Users\Chris>

ivankravets, 2017-05-01:
@orgkhnargh Please remove all Pythons from your virtual machine, then install Python 3.6 and try to install PIO IDE. Clean uninstallation:

- http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

stefan-schindler, 2017-05-01:
@ivankravets I'm not 100% sure, but I think Python 3 has nothing to do with it. I suppose the problem lies in the existing Python 2.7 installation. In some cases it isn't able to properly run the `virtualenv.py` setup script.

closed 2017-05-04 by ivankravets

nevercast, 2017-05-19:
This issue is not fixed. I am using Platform.IO IDE installed from APM _today_ and this is still an issue.

For my case, making `python` point to `python2` instead of `python3` fixed it. 
If you want Python 2.7, I encourage that Platform.IO checks the environment for the presence of `python2` first. That is the correct naming convention for multi-version python systems.

commit 163db66e (2017-05-04): Try to install Python with administrative permissions if “without” fails // Resolve #396 Resolve #337 Resolve #422