platformio/platformio-atom-ide #340: InstallationManager

rianders, 2017-04-06:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/2c/vb3ntrs514q41rqr1zbyy56w0000gr/T/tmp-196181h16UuwC5RuZ/virtualenv-15.1.0/virtualenv.py", line 2328, in <module>

    main()
  File "/var/folders/2c/vb3ntrs514q41rqr1zbyy56w0000gr/T/tmp-196181h16UuwC5RuZ/virtualenv-15.1.0/virtualenv.py", line 713, in main

    symlink=options.symlink)
  File "/var/folders/2c/vb3ntrs514q41rqr1zbyy56w0000gr/T/tmp-196181h16UuwC5RuZ/virtualenv-15.1.0/virtualenv.py", line 945, in create_environment

    download=download,
  File "/var/folders/2c/vb3ntrs514q41rqr1zbyy56w0000gr/T/tmp-196181h16UuwC5RuZ/virtualenv-15.1.0/virtualenv.py", line 901, in install_wheel

    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/var/folders/2c/vb3ntrs514q41rqr1zbyy56w0000gr/T/tmp-196181h16UuwC5RuZ/virtualenv-15.1.0/virtualenv.py", line 797, in call_subprocess

    % (cmd_desc, proc.returncode))
OSError: Command /Users/rk2/.platformio/penv/bin/python2.7 - setuptools pip wheel failed with error code 2

ivankravets, 2017-04-06:
Does Atom restart help you?

rianders, 2017-04-06:
I was asked to restart and then it wanted to update again. Then the error occurred  again.

commit 61d7e67b (2017-04-30): Revert back virtualenv to 14.0.6 // Resolve #276, Resolve #286, Resolve #295, Resolve #340, Resolve #382, Resolve #388, Resolve #391, Resolve #405, Resolve #407, Resolve #411, Resolve #420
closed 2017-04-30 by ivankravets