platformio/platformio-atom-ide #343: InstallationManager

networkfusion, 2017-04-07:
Trying to install the development branch, I am experiencing the following error:

```
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\Robin\AppData\Local\Temp\tmp-213963vjEDtV6Al5g\virtualenv-15.1.0\virtualenv.py", line 2328, in <module>
    main()
  File "C:\Users\Robin\AppData\Local\Temp\tmp-213963vjEDtV6Al5g\virtualenv-15.1.0\virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "C:\Users\Robin\AppData\Local\Temp\tmp-213963vjEDtV6Al5g\virtualenv-15.1.0\virtualenv.py", line 945, in create_environment
    download=download,
  File "C:\Users\Robin\AppData\Local\Temp\tmp-213963vjEDtV6Al5g\virtualenv-15.1.0\virtualenv.py", line 901, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\Robin\AppData\Local\Temp\tmp-213963vjEDtV6Al5g\virtualenv-15.1.0\virtualenv.py", line 753, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Users\Robin\.platformio\python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Users\Robin\.platformio\python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

```

ivankravets, 2017-05-06:
Should be fixed in the latest PIO IDE 2.0 Beta 3

blehke, 2017-07-07:
Getting what I think is the same error (different line numbers, but same files and calls) in 2.0.0-beta.7

ivankravets, 2017-07-07:
@blehke could you try to remove `.platformio` folder from home directory?

blehke, 2017-07-07:
Removed .patformio, reinstalled, same thing.  This was a brand new install of Atom, too (the first time - just before I reported this).

ivankravets, 2017-07-07:
@blehke , please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

blehke, 2017-07-07:
Path:
`C:\Users\blake>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\Python27\;C:\Python27\Scripts;C:\Program Files (x86)\ActiveState Komodo Edit 8\;C:\Program Files\Common Files\Microsoft Shared\Windows Live;C:\Program Files (x86)\Common Files\Microsoft Shared\Windows Live;C:\Program Files (x86)\AMD APP\bin\x86_64;C:\Program Files (x86)\AMD APP\bin\x86;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Dell\DW WLAN Card\Driver;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (x86)\xLights;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Users\blake\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files\nodejs\;C:\Program Files\Microsoft Network Monitor 3\;C:\Program Files (x86)\QuickTime\QTSystem\;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Users\blake\.dnx\bin;C:\Users\blake\AppData\Roaming\Python\Scripts;C:\Users\blake\AppData\Roaming\npm;%USERPROFILE%\AppData\Local\Microsoft\WindowsApps;;C:\Program Files\Docker Toolbox`

Python:
`Python 2.7.8`

pip version:
`pip 1.5.6 from C:\python27\lib\site-packages (pyton 2.7)`

pip list:
`esptool (1.3)
pip (1.5.6)
pypm (1.4.3)
pyserial (3.2.1)
pythonselect (1.3)
pywin32 (218.3)
setuptools (5.2)
vboxapi (1.0)
virtualenv (1.11.6)
wsgiref (0.1.2)`

ivankravets, 2017-07-07:
@blehke please run in `cmd`
```
pip install -U setuptools pip virtualenv
```

Restart Atom

Also, would be good if you update Python to 2.7.13

blehke, 2017-07-07:
I get the feeling I have some other issue... command above had a long log file, which ended with this:
`WindowsError: [Error 5] Access is denied: 'c:\\users\\blake\\appdata\\local\\temp\\pip-vsiize-uninstall\\python27\\scripts\\pip.exe'`

edit: Updated python, re-ran the pip install - looks to have fixed it.

ivankravets, 2017-07-07:
Thanks, the solution was to update Python.

closed 2017-07-07 by ivankravets

amayii0, 2018-12-21:
Hello,
I had a similar issue with PIO 1.33.1 ia32 on Windows.
When upgrading Python from 2.7.13 to 2.7.15, it complained about running processes.
I indeed had around 8-9 python.exe process.
After all of killing them, it complained about process "8196", this was the PID of Googl Drive Sync. After killing it too, upgrade of Python and PIO went fine.
Access denied seems related to background processes.
Just my 2 cents.