platformio/platformio-atom-ide #353: Atmel SAM - Wrong programming utility?

fabltd, 2017-04-10:
Hi 

Atmel SAM appears to be trying to use avrdude to upload code?

Arduino IDE uses Bossa.

Output below:

It doesn't work as is.

------------------------------------

Forcing reset using 1200bps open/close on port COM13
Waiting for the new upload port...
Uploading .pioenvs\mzeroUSB\firmware.hex
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

fabltd, 2017-04-10:
The development version appears to fix the issue:
# install development version
> platformio platform install https://github.com/platformio/platform-atmelsam.git

The release version is broken!

ivankravets, 2017-04-10:
Does development version work for you?

fabltd, 2017-04-10:
Hi

Yes.

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

ivankravets, 2017-04-10:
> Hi
> Yes.

Hm... Arduino IDE uses avrdude for Arduino.Org boards. Are you sure that use M0 board by Arduino.Org?

fabltd, 2017-04-10:
Hi

I am using an Adafruit Feather M0

Jeremy

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

ivankravets, 2017-04-10:
You provided `M0` dump above:
```
Uploading .pioenvs\mzeroUSB
```

> I am using an Adafruit Feather M0

Yes, this board was fixed in dev. We will release development platform this week. Need to apply a few changes related to debugging.

fabltd, 2017-04-10:
Hi

It also complains about ardunio.h missing ?

If I comment out the ardunio include it works?

Regards,
Jeremy Turner

ivankravets, 2017-04-10:
Jeremy, please file new issue and provide all details to reproduce it:

- https://github.com/platformio/platform-atmelsam/issues

closed 2017-04-10 by ivankravets