platformio/platformio-atom-ide #355: Uncaught TypeError: Cannot read property 'headers' of undefined

joonghyung, 2017-04-11:
[Enter steps to reproduce:]

1. I follow this steps.
    http://docs.platformio.org/en/latest/ide/atom.html#installation
    http://docs.platformio.org/en/latest/ide/atom.html#quick-start
2. launch in progress
3. installing project examples / platformIO Core is failed.

**Atom**: 1.15.0 ia32
**Electron**: 1.3.13
**OS**: Unknown Windows version
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.2

### Stack Trace

Uncaught TypeError: Cannot read property 'headers' of undefined

```
At C:\Users\SDS\.atom\packages\platformio-ide\lib\installer\helpers.js:78

TypeError: Cannot read property 'headers' of undefined
    at Request._callback (/packages/platformio-ide/lib/installer/helpers.js:78:32)
    at self.callback (/packages/platformio-ide/node_modules/request/request.js:188:22)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
    at Request.onRequestError (/packages/platformio-ide/node_modules/request/request.js:884:8)
    at emitOne (events.js:96:13)
    at ClientRequest.emit (events.js:188:7)
    at TLSSocket.socketErrorListener (_http_client.js:308:9)
    at emitOne (events.js:96:13)
    at TLSSocket.emit (events.js:188:7)
    at emitErrorNT (net.js:1272:8)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)
```

### Commands

```
     -1:45.2.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
```

### Non-Core Packages

```
autocomplete-clang 0.10.1 
build 0.67.0 
busy 0.7.0 
file-icons 2.1.2 
intentions 1.1.2 
language-ini 1.19.0 
linter 2.1.2 
linter-gcc 0.7.1 
linter-ui-default 1.2.2 
minimap 4.27.1 
platformio-aia 0.0.5 
platformio-ide 2.0.0-beta.2 
platformio-ide-terminal 2.5.0 
tool-bar 1.1.0 
```

-------------------------------------------
InstallationManager

Error: Error: unable to verify the first certificate
    at ProjectExamplesStage.<anonymous> (file:///C:/Users/SDS/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:148:13)
    at throw (<anonymous>)
    at step (C:\Users\SDS\.atom\packages\platformio-ide\lib\installer\stages\project-examples.js:13:273)
    at process._tickCallback (internal/process/next_tick.js:103:7)

Vampir8, 2017-04-21:
I have te same problem

ivankravets, 2017-06-02:
Fixed in the latest beta.4 Please Atom > Settings > Updates

closed 2017-06-02 by ivankravets

polydon, 2018-01-11:
Hi @ivankravets, I have installed the latest version of platformio IDE, but I'm still experiencing this problem.
Any suggestions?
Thanks.

ivankravets, 2018-01-11:
@polydon do you use the latest Atom? Could you provide the full output of error?

polydon, 2018-01-12:
@ivankravets yes I am using Atom 1.23.3
Here is the error report:
[Enter steps to reproduce:]

1. During the installation of Platformio Core

**Atom**: 1.23.3 x64
**Electron**: 1.6.15
**OS**: Microsoft Windows 10 Enterprise 2016 LTSB
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.7

### Stack Trace

Uncaught TypeError: Cannot read property 'headers' of undefined

```
At C:\Users\a5107319\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2737

TypeError: Cannot read property 'headers' of undefined
    at Request._callback (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2737:32)
    at self.callback (/packages/platformio-ide/node_modules/request/request.js:186:22)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
    at Request.onRequestError (/packages/platformio-ide/node_modules/request/request.js:878:8)
    at emitOne (events.js:96:13)
    at ClientRequest.emit (events.js:188:7)
    at TLSSocket.socketErrorListener (_http_client.js:309:9)
    at emitOne (events.js:96:13)
    at TLSSocket.emit (events.js:188:7)
    at emitErrorNT (net.js:1281:8)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)
```

### Commands

```
```

### Non-Core Packages

```
ask-stack 2.2.0 
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.15 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
minimap 4.29.7 
platformio-ide 2.0.0-rc.7 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
tool-bar 1.1.4 
```

and then I get 'unable to verify the first certificate' error similar to the one in the original post:

InstallationManager
Error: Failed to download file https://pypi.python.org/packages/source/v/virtualenv/virtualenv-14.0.6.tar.gz: Error: unable to verify the first certificate
    at _callee$ (C:\Users\a5107319\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2605:19)
    at tryCatch (C:\Users\a5107319\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:828:40)
    at Generator.invoke [as _invoke] (C:\Users\a5107319\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1062:22)
    at Generator.prototype.(anonymous function) [as throw] (C:\Users\a5107319\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:880:21)
    at step (C:\Users\a5107319\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2707:191)
    at C:\Users\a5107319\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2707:402
    at process._tickCallback (internal/process/next_tick.js:103:7)

Thank you for looking into this.

ivankravets, 2018-01-12:
See https://www.google.com.ua/search?client=safari&rls=en&q=nopm+unable+to+verify+the+first+certificate&ie=UTF-8&oe=UTF-8&gfe_rd=cr&dcr=0&ei=w6BYWoKKJZGlX5bDvMAO