platformio/platformio-atom-ide #356: Rebuild spawns an endless stream of Python processes

joscha, 2017-04-11:
* IDE: `2.0.0-beta.2`
* Core: `3.4.0a4`

When rebuilding a project after updating to `2.0.0-beta.2` from `1.7`, the rebuild window spins up an endless number of times, each time setting off an additional Python process that downloads the `atmelsam` framework.

Came across it due to trying out the fix for: https://github.com/platformio/platform-atmelsam/issues/10

[image]

can only get out with:
`killall python2.7` and `killall Atom`

ini looks like:
```ini
[env:sodaq_one]
platform = atmelsam
board = sodaq_one
framework = arduino
build_flags = -D ARDUINO_ARCH_SAMD -D DEBUG
;    -D DISABLE_LOGGING
extra_script = extra_script.py

lib_deps =
    Adafruit ADS1X15@1.0.0
    Sodaq_RN2483@1.0.6
    DHT sensor library@1.3.0
    RTCZero@1.5.1
    RTClib@1.2.0
    Sodaq_UBlox_GPS@0.9.3
    FlashStorage@0.6.0
    Nanopb@0.3.9-dev
    ArduinoLog@1.0.1
    https://github.com/SodaqMoja/SodaqOne-UniversalTracker-v2.git
```

joscha, 2017-04-11:
After a `pio run` on CMD it seems to build, however Atom still seems to think it needs rebuilding:

[image]

joscha, 2017-04-11:
It also seems to re-generate `.travis.yml` in every single invocation.

joscha, 2017-04-11:
Disabling automatic rebuilding seems to help:

[image]

Haven't found out yet what causes it.

joscha, 2017-04-11:
Removing the git-referenced library seems to fix it, e.g.
```ini
lib_deps =
    https://github.com/SodaqMoja/SodaqOne-UniversalTracker-v2.git
```

Changing it to a short notation with fixed sha doesn't seem to make a difference:
```ini
lib_deps =
    SodaqMoja/SodaqOne-UniversalTracker-v2#4b6b52fee10c7f4d31016acb6c34b188cd3a0d71
```

even on command line it seems to re-download/clone the repository every single build run.

joscha, 2017-04-11:
Using a version number, e.g.: `SodaqMoja/SodaqOne-UniversalTracker-v2#v4.4` doesn't seem to change anything either.

ivankravets, 2017-04-30:
Should be fixed in the latest PIO Core. Could you reproduce it again?

ivankravets, 2017-05-06:
Please install the latest PIO IDE 2.0 Beta 3. Should be fixed

closed 2017-05-06 by ivankravets