platformio/platformio-atom-ide #360: InstallationManager pip not recognized as internal or external command

mgulsoy, 2017-04-12:
When trying to install platformio-ide, on the last step named "platformio-core" I got the message (also translated) below and installation stops. I got this error both with x64 and x86 setups.

Here are my system specs:
- Win7 home ed. x64
- Has working installation of python 2.7 and pip.
- Windows ide is Turkish. Region is Turkey. 

PIP: 'C:\Users\mert\.platformio\penv\Scripts\pip' i� ya da d�� komut, �al��t�r�labilir
program ya da toplu i� dosyas� olarak tan�nm�yor.
 ------------------------Translation of the message: 
PIP: 'C:\Users\mert\.platformio\penv\Scripts\pip' not recognized as internal or external command or batch file.

ivankravets, 2017-05-06:
Please open system terminal (`cmd`) and provide:
```
python --version
pip --version
virtualenv -- version
echo %PATH%
```

Thanks!

ivankravets, 2017-05-06:
Also, please update to the latest PIO IDE 2.0 beta 3

closed 2017-05-06 by ivankravets