platformio/platformio-atom-ide #362: Install missed packages: build,platformio-ide-terminal,linter-ui-default,intentions,dbg,dbg-gdb,output-panel

seawayza, 2017-04-13:
Error: EPERM: operation not permitted, unlink 'C:\Users\K\.atom\packages\build'

    at file:///C:/Users/K/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:86:27
    at onExit (file:///C:/Users/K/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/K/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\K\AppData\Local\atom\app-1.14.3\resources\app.asar\src\buffered-process.js:303:11)
    at ChildProcess.<anonymous> (C:\Users\K\AppData\Local\atom\app-1.14.3\resources\app.asar\src\buffered-process.js:333:11)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

ivankravets, 2017-04-30:
Depends on https://github.com/platformio/platformio-atom-ide/issues/424

closed 2017-05-01 by ivankravets

ProPuke, 2017-09-09:
Author of `dbg`, `dbg-gdb` and `output-panel` here.

These packages do appear to be compatible with platformio (now?).

How does one escape the clutches of the naughty (ATOM_CONFLICTED_DEPENDENCIES) list?

ivankravets, 2017-09-09:
@ProPuke Sorry, we had problems before when migrated to our debugging front-end https://github.com/platformio/platformio-atom-ide-debugger

I've just removed these packages https://github.com/platformio/platformio-atom-ide/commit/e32299468dde211f3f80bd8f07416bbc370f69be . So, developers should control which packages to use. Also, need to manually specify debugging configuration for dbg-gdb.

ProPuke, 2017-09-09:
Thanks Ivan.

Your debugger looks nice btw!

commit 9bb32114 (2017-05-01): Install dependencies one by one when failed to install all at once

Resolve #424, Resolve #285, Resolve #288, Resolve #292, Resolve #310,
Resolve #314, Resolve #315, Resolve #331, Resolve #334, Resolve #338,
Resolve #345, Resolve #358, Resolve #362, Resolve #380, Resolve #383,
Resolve #387, Resolve #394, Resolve #397, Resolve #401, Resolve #418,
Resolve #421, Resolve #332, Resolve #339, Resolve #342, Resolve #346,
Resolve #363, Resolve #381, Resolve #386, Resolve #395, Resolve #402