platformio/platformio-atom-ide #369: Docs imply that Library Manager is part of IDE, but there is no gui.

dsjstc, 2017-04-14:
Just spent about half an hour trying to figure out how to use and/or install the Library Manager gui into the IDE My IDE (2.0 beta 2) doesn't match [the docs](http://docs.platformio.org/en/latest/ide/atom.html) in a handful of places.  Specifically, the following imply that there is a Library Manager gui, which one might assume has to be installed separately:

- "Library Manager" is listed in the features for the IDE.  In fact it's a [whole separate element](http://docs.platformio.org/en/latest/librarymanager/index.html)
- [Toolbar](http://docs.platformio.org/en/latest/ide/atom.html#platformio-toolbar) references a button for Library Manager between Serial Monitor and Terminal.

Ultimately, it was [this screenshot](http://docs.platformio.org/en/latest/ide/atom.html#menu-item-platformio) that tipped me off:  it says "gui coming soon", although for some reason that menu item is entirely missing from my IDE.

PS:  This all started because I was looking for an easy way to open an example from an Arduino library as a new project in the IDE -- as of this report, I don't believe there is one.

ivankravets, 2017-04-15:
Have you seen Menu: PlatformIO > Home > Libraries?

dsjstc, 2017-04-15:
Awesome.

It works really well!  However, I don't see it referenced anywhere obvious in the IDE docs.  Having a toolbar button go straight there would probably save a lot of people my grief.

I'll leave the bug open in case you want it as a placeholder for either (a) doc revisions, or (b) restore the toolbar button.   Feel free to close it if you don't.