platformio/platformio-atom-ide #374: InstallationManager, BAD foldername ('á' or 'space') character makes impossible to install.

mzozo95, 2017-04-15:
If the installition folder would be modifyable, it would be great.
I do not understand what platform-io want to do whit FFMPEG...
```
Error: Error: ENOENT: no such file or directory, mkdir 'C:\Users\Majláth Zoltán\AppData\Local\Temp;c:\FFMPEG\bin\tmp-7368IMN1av0tJVUj'
    at ProjectExamplesStage.<anonymous> (file:///C:/Users/Majláth Zoltán/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:148:13)
    at throw (<anonymous>)
    at step (C:\Users\Majláth Zoltán\.atom\packages\platformio-ide\lib\installer\stages\project-examples.js:13:273)
    at process._tickCallback (internal/process/next_tick.js:103:7)
```

mzozo95, 2017-04-15:
Solution: Set windows system environmental variables something like this:
`C:\TEMP\AppDara\Local\Temp`
And the installition can be completed. Set only ONE path, the atom can not understand the split ';' character!

ivankravets, 2017-04-30:
Will be resolved in this issue https://github.com/platformio/platformio-atom-ide/issues/321

closed 2017-04-30 by ivankravets