platformio/platformio-atom-ide #392: InstallationManager

Jbbs, 2017-04-21:
Virtualenv Download: Traceback (most recent call last):
  File "c:\temp\tmp-31388JaFIXf4FE0FM\virtualenv-15.1.0\virtualenv.py", line 2328, in <module>
    main()
  File "c:\temp\tmp-31388JaFIXf4FE0FM\virtualenv-15.1.0\virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "c:\temp\tmp-31388JaFIXf4FE0FM\virtualenv-15.1.0\virtualenv.py", line 945, in create_environment
    download=download,
  File "c:\temp\tmp-31388JaFIXf4FE0FM\virtualenv-15.1.0\virtualenv.py", line 901, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "c:\temp\tmp-31388JaFIXf4FE0FM\virtualenv-15.1.0\virtualenv.py", line 753, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Users\401993\.platformio\python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Users\401993\.platformio\python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Accesso negato

Jbbs, 2017-04-21:
This happens during the platformIO core installation launced from ATOM, both as a normal user that as admin

ivankravets, 2017-04-30:
Do you have Python installed in a system?

Jbbs, 2017-05-01:
Yes,
I think that is something proxy related. I'm on windows 7 and I had the error at office, where I must  use a proxy server, but I was able to install without problems at home without proxy.
The error persist also if I define HTTP PROXYs environment variables.
At home I installed also un linux Ubunto without problems.

ivankravets, 2017-05-06:
Could you try to update to the latest PIO IDE 2.0 Beta 3? Do you use the latest Atom v1.16?

closed 2017-06-01 by ivankravets