platformio/platformio-atom-ide #408: Failed to run BUILD command afert "Try PlatformeIO IDE 2.0 Preview"

philuser, 2017-04-26:
Unable to launch a build after upgrading the PlatformIO IDE 2.0
I systemetically the error:

Failed to run BUILD command

I go back to IDE 1.7.3 | Core 3.3.0
After this back is correct again good

I tried again with an install from scratch after deleting the directories "~ / .atom" and "~ / .platformio". The result is identical, the lasted "platformio 2.0 preview" does not accept to launch a build

I can tell you other things, if this can help you solve this problem

my OS : Linux OpenSuse Tumbleweed

ivankravets, 2017-04-26:
Is your project symlinked?

philuser, 2017-04-28:
I think no, but I would do a new test, this weekend, with a project at the root of my account.  I keep you informed.

ivankravets, 2017-04-30:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/320

closed 2017-04-30 by ivankravets

philuser, 2017-05-01:
There were no symbolic links to my projects.
I therefore recommenced my tests as follows
  - deletion on the laptop of directories ~/.platformio and ~/.atom, deletion and instalation of atom and PlatformIO IDE 2.0.0-beta.2 -> Failed to build
  - Installing on another computer with the same OS "Opensuse Tumbleweed" Atom + Platformio IDE 2.0.0-beta.2. Creates a shared NFS directory from my laptop on the same root of my projects -> Perfect everything works

__Conclusion :__
Nothing to do with the OS, nor with the tree of my projects, but certainly with a residual config of Platformio, that I did not find!
An idea ?

For the moment I have switched back on a version 1.7.3 of Plaformio on my laptop

philuser, 2017-05-01:
Hi, I just discovered the existence of a beta-3, but not yet available in the update available in atom-platformio.
I would make a test of the availability of this new patch

Thank you very much, with my encouragement for your work