platformio/platformio-atom-ide #423: InstallationManager

igbor, 2017-04-30:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\igbor\AppData\Local\Temp\tmp-25952Y5zQe7ISmTvw\virtualenv-15.1.0\virtualenv.py", line 2328, in <module>
    main()
  File "C:\Users\igbor\AppData\Local\Temp\tmp-25952Y5zQe7ISmTvw\virtualenv-15.1.0\virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "C:\Users\igbor\AppData\Local\Temp\tmp-25952Y5zQe7ISmTvw\virtualenv-15.1.0\virtualenv.py", line 945, in create_environment
    download=download,
  File "C:\Users\igbor\AppData\Local\Temp\tmp-25952Y5zQe7ISmTvw\virtualenv-15.1.0\virtualenv.py", line 901, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\igbor\AppData\Local\Temp\tmp-25952Y5zQe7ISmTvw\virtualenv-15.1.0\virtualenv.py", line 797, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\igbor\.plat...v\Scripts\python.exe - setuptools pip wheel failed with error code 2

ivankravets, 2017-04-30:
Could you try in Windows Terminal
```
apm install platformio/platformio-atom-ide
```

Be sure that GIT is installed in your system. Thanks.

ivankravets, 2017-05-06:
Please open system terminal (`cmd`) and provide:
```
python --version
pip --version
virtualenv -- version
echo %PATH%
```

Thanks!

ivankravets, 2017-06-27:
Please reopen if you still need help

closed 2017-06-27 by ivankravets

me21, 2017-10-27:
It was fixed for me by installing virtualenv package in Python manually.