platformio/platformio-atom-ide #425: InstallationManager

g0ttl, 2017-05-01:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-9328AFHuIYHNi0YJ\virtualenv-15.1.0\virtualenv.py", line 2328, in <module>
    main()
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-9328AFHuIYHNi0YJ\virtualenv-15.1.0\virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-9328AFHuIYHNi0YJ\virtualenv-15.1.0\virtualenv.py", line 945, in create_environment
    download=download,
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-9328AFHuIYHNi0YJ\virtualenv-15.1.0\virtualenv.py", line 901, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-9328AFHuIYHNi0YJ\virtualenv-15.1.0\virtualenv.py", line 797, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\g0ttl\.plat...v\Scripts\python.exe - setuptools pip wheel failed with error code 2

ivankravets, 2017-05-01:
Could you try the latest development build? Close Atom and open Windows `cmd` and paste:

```
apm install platformio/platformio-atom-ide --production
```

Then, please start Atom.

g0ttl, 2017-05-01:
I try you cmd, but have a new problem
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-45367a4lFdLADsFB\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-45367a4lFdLADsFB\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-45367a4lFdLADsFB\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-45367a4lFdLADsFB\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\g0ttl\AppData\Local\Temp\tmp-45367a4lFdLADsFB\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\g0ttl\.plat...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

g0ttl, 2017-05-01:
other, i install a Clang in my system and  

>  Please select “Add LLVM to the system PATH” option on the installation step.

but Atom also hints 

> Clang is not installed in your system!

ivankravets, 2017-05-01:
What is your OS?

g0ttl, 2017-05-02:
Edition: Windows 10
Version: 1607
OS Build: 14393.1066
System type:64-bit

ivankravets, 2017-05-06:
Please open system terminal (`cmd`) and provide:
```
python --version
pip --version
echo %PATH%
```

Thanks!

g0ttl, 2017-05-11:
[image]

g0ttl, 2017-05-11:
[image]
Microsoft Windows [Version 10.0.14393]
(c) 2016 Microsoft Corporation. All rights reserved.

C:\Users\g0ttl>python --version
Python 3.5.2

C:\Users\g0ttl>pip --version
pip 9.0.1 from c:\users\g0ttl\appdata\local\programs\python\python35\lib\site-packages (python 3.5)

C:\Users\g0ttl>echo %PATH%
C:\Perl64\site\bin;C:\Perl64\bin;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Windows\SysWOW64\WindowsPowerShell\v1.0\Modules\TShell\TShell\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\Microsoft Network Monitor 3\;C:\Program Files (x86)\Notepad++;C:\Program Files\Java\jdk1.8.0_101\bin;C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\amd64;C:\Program Files\Git\cmd;D:\Program Files\nodejs\;D:\wamp64\bin\mysql\mysql5.7.14\bin;D:\Program Files\PuTTY\;C:\Program Files\LLVM\bin;C:\Users\g0ttl\AppData\Local\Programs\Python\Python35\Scripts\;C:\Users\g0ttl\AppData\Local\Programs\Python\Python35\;C:\Users\g0ttl\AppData\Local\Microsoft\WindowsApps;C:\Users\g0ttl\AppData\Local\atom\bin;%SHPLAYER%;C:\Users\g0ttl\AppData\Roaming\npm;D:\Program Files\nodejs\node_global;C:\Program Files (x86)\Microsoft VS Code\bin

C:\Users\g0ttl>

ivankravets, 2017-06-01:
Do you see `C:/Users/g0ttl/.platformio/python27` folder here?

ivankravets, 2017-06-27:
Please reopen if you still need help

closed 2017-06-27 by ivankravets

me21, 2017-10-27:
@g0ttl they say Python 3 is not supported at the moment.

I have the same error, tried to install Platformio IDE in Atom and Visual Studio Code.

Python version: 2.7.14
Pip version: 9.0.1 from c:\python27\lib\site-packages (python 2.7)
Path: C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\OpenCL SDK\2.0\bin\x86;C:\Program Files (x86)\Intel\OpenCL SDK\2.0\bin\x64;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\LLVM\bin;C:\Users\fgka\AppData\Local\Microsoft\WindowsApps;C:\Python27;;C:\Program Files\Microsoft VS Code\bin

me21, 2017-10-27:
It was fixed for me by installing virtualenv package in Python manually.