platformio/platformio-atom-ide #433: InstallationManager

aljsr, 2017-05-03:
PIP: Traceback (most recent call last):
  File "C:\Python27\Lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\Python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\alan.ABERLINK\.platformio\penv\Scripts\pip.exe\__main__.py", line 5, in <module>
  File "c:\users\alan.aberlink\.platformio\penv\lib\site-packages\pip\__init__.py", line 21, in <module>
    from pip._vendor.requests.packages.urllib3.exceptions import DependencyWarning
  File "c:\users\alan.aberlink\.platformio\penv\lib\site-packages\pip\_vendor\requests\__init__.py", line 62, in <module>
    from .packages.urllib3.exceptions import DependencyWarning
  File "c:\users\alan.aberlink\.platformio\penv\lib\site-packages\pip\_vendor\requests\packages\__init__.py", line 29, in <module>
    import urllib3
ImportError: No module named urllib3

ivankravets, 2017-05-06:
Could you open system terminal (`cmd`) and type
```
python --version
pip --version
virtualenv --version
```

Thanks!

ivankravets, 2017-06-01:
Should be fixed in the latest python requests

closed 2017-06-01 by ivankravets