platformio/platformio-atom-ide #441: EnvironmentError: No tool named 'pioide': not a Zip file

davidcranor, 2017-05-05:
I was experiencing an issue where after a hung install (I'm in China and internet here is sketchy), my IDE would hang on the updating core dialog.  Fixed it by deleting `/home/<user>/.atom/packages/platformio-ide/` and reinstalling from the Atom package manager.  But now, when I attempt to build a project, I get the following error.  Any ideas as to what is going on?   Thanks!

[Fri May  5 18:23:47 2017] Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------
EnvironmentError: No tool named 'pioide': not a Zip file:
File "/Users/davidcranor/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 105:
env = DefaultEnvironment(**DEFAULT_ENV_OPTIONS)
File "/Users/davidcranor/.platformio/packages/tool-scons/script/../engine/SCons/Defaults.py", line 88:
_default_env = SCons.Environment.Environment(*args, **kw)
File "/Users/davidcranor/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 982:

apply_tools(self, tools, toolpath)
File "/Users/davidcranor/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 107:
env.Tool(tool)
File "/Users/davidcranor/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 1788:
tool = SCons.Tool.Tool(tool, toolpath, **kw)
File "/Users/davidcranor/.platformio/packages/tool-scons/script/../engine/SCons/Tool/__init__.py", line 104:
module = self._tool_module()
File "/Users/davidcranor/.platformio/packages/tool-scons/script/../engine/SCons/Tool/__init__.py", line 164:
raise SCons.Errors.EnvironmentError(m)
========================== [ERROR] Took 0.63 seconds ==========================

ivankravets, 2017-05-06:
Please remove this folder `/Users/davidcranor/.platformio/penv` and restart Atom.

closed 2017-05-06 by ivankravets

davidcranor, 2017-05-07:
Still didn't work....

It reinstalled the core, and penv dir is back in my .platformio directory, but still getting the same error.  Also, it gives me the same error when I try to initialize a new project.

Making a platformio project and building it using pio/platformio init and run from the command line seem to work fine, though.

ivankravets, 2017-05-07:
It looks that you have VERY OLD PIO Core installed in a system. Please open SYSTEM terminal (not in IDE) and run `pip uninstall platformio`. Then it should work from PIO IDE Beta 3

davidcranor, 2017-05-09:
That worked, thank you!