platformio/platformio-atom-ide #446: InstallationManager

enricotimisela, 2017-05-05:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\etimisel\AppData\Local\Temp\tmp-1488VP0m0ooTDEhV\virtualenv-15.1.0\virtualenv.py", line 2328, in <module>
    main()
  File "C:\Users\etimisel\AppData\Local\Temp\tmp-1488VP0m0ooTDEhV\virtualenv-15.1.0\virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "C:\Users\etimisel\AppData\Local\Temp\tmp-1488VP0m0ooTDEhV\virtualenv-15.1.0\virtualenv.py", line 945, in create_environment
    download=download,
  File "C:\Users\etimisel\AppData\Local\Temp\tmp-1488VP0m0ooTDEhV\virtualenv-15.1.0\virtualenv.py", line 901, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\etimisel\AppData\Local\Temp\tmp-1488VP0m0ooTDEhV\virtualenv-15.1.0\virtualenv.py", line 797, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\etimisel\.p...v\Scripts\python.exe - setuptools pip wheel failed with error code 2

ivankravets, 2017-05-06:
Could you uninstall PIO IDE http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide and install the latest Beta 3?

Thanks!

ivankravets, 2017-05-18:
Please open system terminal (`cmd`) and provide:
```
python --version
pip --version
virtualenv -- version
echo %PATH%
```

Thanks!

closed 2017-06-01 by ivankravets