platformio/platformio-atom-ide #447: Update to Platformio 2.0.0 beta3 keeps reinstalling

chessweb01, 2017-05-06:
Today I updated Platformio IDE to 2.0.0 beta3 on Windows 7 64 bit. After the required Atom restart, the system once again started to download and install PlatformIO core and required a restart upon completion. This cycle now repeats with every start of Atom.

I unsuccessfully tried to fix it by uninstalling everything PlatformIO related, closing Atom and reinstalling PlatformIO IDE. The download/install cycle of PlatformIO core still happens with every Atom start which is rather annoying.

ivankravets, 2017-05-06:
1. Do you have PIO Core installed in a system globally? Please open system terminal (`cmd`) and type `pio --version`.
2. Have you installed Python before PIO IDE?

chessweb01, 2017-05-06:
1. PlatformIO, version 3.3.0
2. Yes

ivankravets, 2017-05-06:
PIO IDE Beta3 depends on new PIO Core 3.4.beta1. It adds PIO Unified Debugger feature to IDE.

1. Please uninstall global PIO Core from a system: `pip uninstall platformio`
2. PIO IDE: Menu > PlatformIO > Settings > PIO IDE > Check "Use built-in PIO Core"

chessweb01, 2017-05-06:
That solved the problem, but now I get an error when trying to build a project. Here is the complete report:

Please wait while upgrading PlatformIO...
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.5.0          [Up-to-date]
Updating toolchain-atmelavr              @ 1.40902.0      [Up-to-date]
 

Platform Espressif 8266
--------
Updating espressif8266                   @ 1.3.0          [Up-to-date]
Updating tool-mkspiffs                   @ 1.102.0        [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating tool-espotapy                   @ 1.0.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20300.1      [Skip]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
 
Platform Espressif 8266 (Stage)
--------
Updating espressif8266_stage             @ cbfb942        [Up-to-date]
Updating tool-mkspiffs                   @ 1.102.0        [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating tool-espotapy                   @ 1.0.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20400.0-alpha.2[Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
 
Platform Teensy
--------

Updating teensy                          @ 2.0.0          [Up-to-date]
Updating toolchain-atmelavr              @ 1.40902.0      [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.40804.0      [Up-to-date]
Uninstalling framework-arduinoteensy @ 1.135.0:         Error: Traceback (most recent call last):
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\__main__.py", line 106, in main
    cli(None, None, None)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\click\core.py", line 1024, in invoke
    Command.invoke(self, ctx)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
PIO Buildsrc\ads_teensy.cpp000118:1(1, 27)
 go-plusCRLFUTF-8C++master+1, -1
    return callback(*args, **kwargs)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\__main__.py", line 75, in cli
    maintenance.on_platformio_start(ctx, force, caller)

  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 66, in on_platformio_start
    after_upgrade(ctx)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 173, in after_upgrade
    if u.run(ctx):
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 111, in run
    result.append(callback(ctx))
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 154, in _update_dev_platforms
    ctx.invoke(cmd_platform_update)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 373, in platform_update
    platform, only_packages=only_packages, only_check=only_check)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 124, in update
    self.cleanup_packages(p.packages.keys())
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 149, in cleanup_packages
   pm.uninstall(manifest['__pkg_dir'], trigger_event=False)
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 670, in uninstall
    util.rmtree_(pkg_dir)

  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\util.py", line 592, in rmtree_
    return rmtree(path, onerror=_onerror)
  File "C:\Python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\Lib\shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info())
  File "c:\users\rsel\.platformio\penv\lib\site-packages\platformio\util.py", line 590, in _onerror
    os.remove(name)
WindowsError: [Error 5] Access denied: 'c:\\Users\\rsel\\.platformio\\packages\\framework-arduinoteensy\\libraries\\NXPMotionSense
\\examples'

ivankravets, 2017-05-06:
I don't know how do you run Atom and other Apps. It looks that you have used different permissions when used PIO Core (administrative). Now, of course, you use Atom without Administrative rights but previous PIO Core file structure was created via that user.

The easy solution is to close Atom and remove `c:\Users\rsel\.platformio` folder totally. Atom using PIO Core will create own PIO Core storage with right permissions.

Next time if you need PIO Core without IDE, just add `c:\Users\rsel\.platformio\penv\Scripts` folder to your system env `PATH` and `pio` command will be available in `cmd`.

chessweb01, 2017-05-06:
I tried to build again and noticed that the folder causing the access denied error changed from run to run until the error disappeared altogether. Everything seems to be working now.

ivankravets, 2017-05-06:
Could we close this issue?

chessweb01, 2017-05-06:
Yes. Thanks for the quick help.

ivankravets, 2017-05-06:
Happy Coding with PlatformIO!

closed 2017-05-06 by ivankravets