platformio/platformio-atom-ide #448: InstallationManager

mddypaul, 2017-05-07:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\Manas\AppData\Local\Temp\tmp-7836CTHv2GHO2BAE\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Manas\AppData\Local\Temp\tmp-7836CTHv2GHO2BAE\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Manas\AppData\Local\Temp\tmp-7836CTHv2GHO2BAE\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\Manas\AppData\Local\Temp\tmp-7836CTHv2GHO2BAE\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\Manas\AppData\Local\Temp\tmp-7836CTHv2GHO2BAE\virtualenv-14.0.6\virtualenv.py", line 743, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Python27\Lib\subprocess.py", line 710, in __init__
    errread, errwrite)
  File "C:\Python27\Lib\subprocess.py", line 958, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

ivankravets, 2017-05-07:
Could open system terminal `cmd` and type
```
python --version
pip --version
```

Please provide an output. Thanks.

ivankravets, 2017-06-01:
Should be fixed in 2.0.0.beta.4. Please upgrade

closed 2017-06-01 by ivankravets