platformio/platformio-atom-ide #450: Failed to reset password.

jkDesignDE, 2017-05-08:
Error
    at file:///C:/Users/jochen.krapf/.atom/packages/platformio-ide/lib/account/helpers.js:30:25
    at onExit (file:///C:/Users/jochen.krapf/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/jochen.krapf/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\jochen.krapf\AppData\Local\atom\app-1.16.0\resources\app.asar\src\buffered-process.js:322:11)
    at ChildProcess.<anonymous> (C:\Users\jochen.krapf\AppData\Local\atom\app-1.16.0\resources\app.asar\src\buffered-process.js:352:11)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

orgkhnargh, 2017-05-08:
Hi @jkDesignDE,

Can you please try to reset password via the CLI and let me know what error it prints (if any)? Run the following command in a terminal inside Atom:

```
pio account forgot --username {your email}
```

jkDesignDE, 2017-05-08:
typed in the command:
No response = no error!?!

orgkhnargh, 2017-05-08:
Doesn't sound right.

`pio --version`?

jkDesignDE, 2017-05-08:
pio --version ->
PlatformIO, version 3.4.0b2

orgkhnargh, 2017-05-08:
I wasn't able to reproduce this error using PlatformIO IDE 2.0.0.beta3 and PlatformIO Core 3.4.0b2 on Windows 7.

Can you please reinstall the IDE package and let me know if you still see the error?

closed 2017-06-01 by ivankravets