platformio/platformio-atom-ide #452: TypeError: Cannot read property '0' of undefined

BBaysinger, 2017-05-08:
I'm getting this error in Angular 2 projects. Not sure if it happens in other projects. I'm seeing this on OS X 10.12.4 and Atom 1.16.0. I also saw it in Atom 1.13.0.

```
TypeError: Cannot read property '0' of undefined
    at Rule.SelectorRule [as constructor] (/Users/bradley/Work/pushgraph-client/pushgraph-client/node_modules/codelyzer/selectorNameBase.js:17:24)
    at new Rule (/Users/bradley/Work/pushgraph-client/pushgraph-client/node_modules/codelyzer/directiveSelectorRule.js:12:28)
    at Object.loadRules (/Users/bradley/.atom/packages/linter-tslint/node_modules/tslint/lib/ruleLoader.js:47:36)
    at Linter.getEnabledRules (/Users/bradley/.atom/packages/linter-tslint/node_modules/tslint/lib/linter.js:168:44)
    at Linter.lint (/Users/bradley/.atom/packages/linter-tslint/node_modules/tslint/lib/linter.js:81:33)
    at /Users/bradley/.atom/packages/linter-tslint/lib/main.js:149:18
```

orgkhnargh, 2017-05-09:
None of the PlatformIO IDE files is mentioned in the traceback, so I'm not sure if this issue is related to PlatformIO IDE at all.

My guess is that `directive-selector` rule isn't configured correctly in your tslint config.

ivankravets, 2017-05-09:
PIO IDE 2.0 adds global error listener. It seems that error was caught from other npm dependency.

closed 2017-05-09 by ivankravets