platformio/platformio-atom-ide #453: Annoying signup pop up at start !

danielbibit, 2017-05-09:
After the last update, every time that I start Atom, a platformio popup comes up asking me to sign up. Apparently there's no configuration for disable this behavior, and is getting really annoying.

ivankravets, 2017-05-09:
Please Menu: PlatformIO > Update PIO packages, libraries. It's is fixed in PIO Core+ 0.8.10

Then logout and login again. 

Please sorry

ghs, 2017-05-09:
@ivankravets it is still happening here even after updating everything.

[image]
[image]

ivankravets, 2017-05-09:
@orgkhnargh Please add a new item to IDE Settings after `showPIOHome` called `showPIOLogin`, "Show PIO Account Login on startup", which should be activated by default.

Please check this setting before showing PIO Account Login window.

ivankravets, 2017-05-09:
@ghs which theme do you use?

ghs, 2017-05-09:
Seti

PR #456 (2017-05-09): Add an option to disable auth check on IDE startup
Resolve #453
closed 2017-05-09 by ivankravets