platformio/platformio-atom-ide #455: InstallationManager installing 2.0.0-beta.3 fails on Mac OS X 10.12

axello, 2017-05-09:
Might be related to #446, but I'm not a python dev. Also different OS. I could not find a non-beta version of the platform-ide. Is this on purpose?

Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-94209sErEzw1q9xXw/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-94209sErEzw1q9xXw/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-94209sErEzw1q9xXw/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment

    download=download,
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-94209sErEzw1q9xXw/virtualenv-14.0.6/virtualenv.py", line 877, in install_wheel

    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-94209sErEzw1q9xXw/virtualenv-14.0.6/virtualenv.py", line 783, in call_subprocess

    % (cmd_desc, proc.returncode))
OSError: Command /Users/axel/.platformio/penv/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

axello, 2017-05-09:
Maybe this is relevant: I have
pip 9.0.1 from /Library/Python/2.7/site-packages (python 2.7)

ivankravets, 2017-05-09:
Could you open system terminal and run `sudo pip install virtualenv` and restart Atom. Does it work now?

axello, 2017-05-10:
Hi,

I did this, and after starting up ATOM it automagically continued installing platformio-ide. But it failed with same errors.

Interestingly, sudo pip install virtualenv returns the following error:
The directory '/Users/axel/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.

However:
imek:pip axel$ whoami
	axel
imek:pip axel$ pwd
	/Users/axel/Library/Caches/pip
imek:pip axel$ ls -al
total 8
drwx------    5 axel  staff   170 Apr 26 10:07 .
drwx------+ 172 axel  staff  5848 May 10 10:09 ..
drwx------   15 axel  staff   510 May  9 11:23 http
-rw-r--r--    1 axel  staff   121 Apr 26 10:07 selfcheck.json
drwxr-xr-x    3 axel  staff   102 Mar 20 14:25 wheels

So a bit of a weird error there. 

After starting up ATOM again, the errors were gone and platformio-ide seems to have installed itself. Oh no: it still continues to install the last package and fail. 
- Uninstalling all platformio-*
- Quit Atom,
- Start + Installing platform-ide 2.0.0-beta3 again: same error
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-134184K75uxE7sKLN/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-134184K75uxE7sKLN/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-134184K75uxE7sKLN/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment
    download=download,
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-134184K75uxE7sKLN/virtualenv-14.0.6/virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-134184K75uxE7sKLN/virtualenv-14.0.6/virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /Users/axel/.platformio/penv/bin/python -c “import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

—————
I must say, I DO have a different version of virtualenv:
pip list  | grep virtual
—> virtualenv (15.1.0)

I then tried:
sudo -H pip install virtualenv 14.0.6
—> Requirement already satisfied: virtualenv in /Library/Python/2.7/site-packages
Collecting 14.0.6
  Could not find a version that satisfies the requirement 14.0.6 (from versions: )
No matching distribution found for 14.0.6

Hope this helps,

Axel

> On 9 May 2017, at 22:25, Ivan Kravets <notifications@github.com> wrote:
> 
> sudo pip install virtualenv

orgkhnargh, 2017-05-10:
@axello 

> Interestingly, sudo pip install virtualenv returns the following error:
> The directory '/Users/axel/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache [...]

This is irrelevant, unless you do python development and want python wheels to be cached to speed up later installations. You can either ignore this warning or use the `-H` flag without any significant issues.

> I then tried: 
> sudo -H pip install virtualenv 14.0.6

`pip` recognized `14.0.6` as a second package name to install instead of a version of the first package. Correct way to specify a version is to use `==` like this:  `sudo -H pip install virtualenv==14.0.6`.

axello, 2017-05-10:
Thanks Dmytro,

Okay, here is what I did this time:

sudo -H pip install virtualenv==14.0.6
Collecting virtualenv==14.0.6
  Downloading virtualenv-14.0.6-py2.py3-none-any.whl (1.8MB)
    100% |████████████████████████████████| 1.8MB 133kB/s 
Installing collected packages: virtualenv
  Found existing installation: virtualenv 15.1.0
    Uninstalling virtualenv-15.1.0:
      Successfully uninstalled virtualenv-15.1.0
Successfully installed virtualenv-14.0.6

- Removed platformio-* from Atom
- Quit Atom, 
- Restart Atom

Exactly same or similar error message:

Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-15543O1Vp9K7hIpLv/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-15543O1Vp9K7hIpLv/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-15543O1Vp9K7hIpLv/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/axel/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath
    if not isabs(path):
  File "/Users/axel/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

Cheers,

Axel

> On 10 May 2017, at 10:49, Dmytro Kyrychuk <notifications@github.com> wrote:
> 
> sudo -H pip install virtualenv==14.0.6.

orgkhnargh, 2017-05-10:
The error is definitely different. It appears to be the same error as in #295, but in that  case it was fine after a restart.

I think it still doesn't use your system virtualenv (the one that should have been installed by `sudo -H pip install virtualenv==14.0.6`). Can you please repeat the process and afterwards open the Developer Tools (View > Developer > Toggle Developer Tools), switch to Console tab there, see what errors precede this `AttributeError: 'NoneType' object has no attribute 'startswith' ` and post them here (if any)?

Thank you.

axello, 2017-05-10:
Here you go, I hope it is complete.

IMHO, this looks weird, but as I said: I’m not a python dev.
OSError: Command /Users/axel/.platformio/penv/bin/python -c “import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 

Console output:

Error: Missed packages: platformio-ide-terminal,platformio-ide-debugger(…)
utils.js [sm]:65 runCommand platformio ["--version"] Object {cacheValid: "10s"}
manager.js [sm]:79 Error: spawn platformio ENOENT
utils.js [sm]:65 runCommand conda ["--version"] Object {}
utils.js [sm]:65 runCommand /Applications/Tools/Development/Atom.app/Contents/Resources/app/apm/bin/apm ["--no-color", "install", "platformio-ide-terminal", "platformio-ide-debugger", "--production", "--compatible"] Object {}
utils.js [sm]:65 runCommand /Users/axel/.platformio/penv/bin/python ["-c", "import sys; print '.'.join(str(v) for v in sys.version_info[:2])"] Object {}
utils.js [sm]:65 runCommand python2.7 ["-c", "import sys; print '.'.join(str(v) for v in sys.version_info[:2])"] Object {}
utils.js [sm]:65 runCommand virtualenv ["-p", "python2.7", "/Users/axel/.platformio/penv"] Object {}
platformio-core.js [sm]:370 User's Virtualenv: Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 2304, in <module>

    main()
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 924, in create_environment

    download=download,
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 877, in install_wheel

    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 783, in call_subprocess

    % (cmd_desc, proc.returncode))

OSError: Command /Users/axel/.platformio/penv/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2
(anonymous function) @ platformio-core.js [sm]:370step @ platformio-core.js:13
utils.js [sm]:65 runCommand /Applications/Tools/Development/Atom.app/Contents/Resources/app/apm/bin/apm ["--no-color", "config", "get", "https-proxy"] Object {cacheValid: "1h"}
utils.js [sm]:65 runCommand python2.7 ["/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/t…16350lA86XkQ1fjyD/virtualenv-14.0.6/virtualenv.py", "/Users/axel/.platformio/penv"] Object {}__proto__: Object__defineGetter__: __defineGetter__()__defineSetter__: __defineSetter__()__lookupGetter__: __lookupGetter__()__lookupSetter__: __lookupSetter__()arguments: nullcaller: nulllength: 1name: "__lookupSetter__"__proto__: ()<function scope>constructor: Object()hasOwnProperty: hasOwnProperty()isPrototypeOf: isPrototypeOf()arguments: nullcaller: nulllength: 1name: "isPrototypeOf"__proto__: ()<function scope>propertyIsEnumerable: propertyIsEnumerable()toLocaleString: toLocaleString()toString: toString()valueOf: valueOf()get __proto__: __proto__()set __proto__: __proto__()
utils.js [sm]:41 InstallationManager Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-16350lA86XkQ1fjyD/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-16350lA86XkQ1fjyD/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/0h/h4b11_ns2bz_yjnktbhxmt940000gn/T/tmp-16350lA86XkQ1fjyD/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/axel/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/Users/axel/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs

    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'
notifyError @ utils.js [sm]:41(anonymous function) @ main.js [sm]:81step @ main.js:11
undefined:1 Uncaught (in promise) undefined
atom-dependencies.js [sm]:100 Installing platformio-ide-terminal to /Users/axel/.atom/packages ✓

Installing platformio-ide-debugger to /Users/axel/.atom/packages ✓

orgkhnargh, 2017-05-10:
This error may happen when trying to create an env inside of an existing env using that existing env python interpreter, [I think](https://github.com/pypa/virtualenv/blob/14.0.6/virtualenv.py#L1061-L1063). Hypothetically, in the IDE this may happen when env already exists but the installer cannot remove it in order to create a new one (this happens when PlatformIO Core is not installed, of when its version is not compatible with the current IDE), and then the installer would try to find an interpreter to make an env from and find the one from the existing env.

@axello, can you verify if `~/.platformio/penv` directory exists, and if it does also verify if you can remove it without sudo (e.g., with `rm -r ~/.platformio/penv`)?

`ls -la ~/.platformio` might be useful too.

axello, 2017-05-23:
Sorry for the late response; busy on other projects. 
~/.platformio/penv exists and is owned by me all the way through.
`imek:.platformio axel$ ls -alh
total 0
drwxr-xr-x    4 axel  staff   136B May 11 10:34 .
drwxr-xr-x+ 136 axel  staff   4.5K May 23 16:44 ..
drwxr-xr-x    6 axel  staff   204B May 11 10:34 penv
drwxr-xr-x   26 axel  staff   884B May  9 11:37 project-examples
`
Except that penv/include contains a reference to `/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7`

and ~/.platformio/penv/lib/python2.7 contains mostly refs to system libraries:
`lrwxr-xr-x  1 axel  staff    82B May 11 10:34 UserDict.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/UserDict.py
-rw-r--r--  1 axel  staff   9.6K May 11 10:34 UserDict.pyc
lrwxr-xr-x  1 axel  staff    81B May 11 10:34 _abcoll.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/_abcoll.py
-rw-r--r--  1 axel  staff    27K May 11 10:34 _abcoll.pyc
lrwxr-xr-x  1 axel  staff    85B May 11 10:34 _weakrefset.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/_weakrefset.py
-rw-r--r--  1 axel  staff    10K May 11 10:34 _weakrefset.pyc
lrwxr-xr-x  1 axel  staff    77B May 11 10:34 abc.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/abc.py
-rw-r--r--  1 axel  staff   6.2K May 11 10:34 abc.pyc
lrwxr-xr-x  1 axel  staff    80B May 11 10:34 codecs.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/codecs.py
-rw-r--r--  1 axel  staff    38K May 11 10:34 codecs.pyc
lrwxr-xr-x  1 axel  staff    77B May 11 10:34 config -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/config
lrwxr-xr-x  1 axel  staff    82B May 11 10:34 copy_reg.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/copy_reg.py
-rw-r--r--  1 axel  staff   5.2K May 11 10:34 copy_reg.pyc
drwxr-xr-x  5 axel  staff   170B May 11 10:34 distutils
lrwxr-xr-x  1 axel  staff    80B May 11 10:34 encodings -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/encodings
lrwxr-xr-x  1 axel  staff    81B May 11 10:34 fnmatch.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/fnmatch.py
-rw-r--r--  1 axel  staff   3.7K May 11 10:34 fnmatch.pyc
lrwxr-xr-x  1 axel  staff    85B May 11 10:34 genericpath.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/genericpath.py
-rw-r--r--  1 axel  staff   3.7K May 11 10:34 genericpath.pyc
lrwxr-xr-x  1 axel  staff    82B May 11 10:34 lib-dynload -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload
lrwxr-xr-x  1 axel  staff    83B May 11 10:34 linecache.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/linecache.py
-rw-r--r--  1 axel  staff   3.3K May 11 10:34 linecache.pyc
lrwxr-xr-x  1 axel  staff    80B May 11 10:34 locale.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/locale.py
-rw-r--r--  1 axel  staff    54K May 11 10:34 locale.pyc
-rw-r--r--  1 axel  staff     0B May 11 10:34 no-global-site-packages.txt
lrwxr-xr-x  1 axel  staff    80B May 11 10:34 ntpath.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ntpath.py
-rw-r--r--  1 axel  staff    56B May 11 10:34 orig-prefix.txt
lrwxr-xr-x  1 axel  staff    76B May 11 10:34 os.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/os.py
-rw-r--r--  1 axel  staff    26K May 11 10:34 os.pyc
lrwxr-xr-x  1 axel  staff    83B May 11 10:34 posixpath.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/posixpath.py
-rw-r--r--  1 axel  staff    12K May 11 10:34 posixpath.pyc
lrwxr-xr-x  1 axel  staff    76B May 11 10:34 re.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/re.py
-rw-r--r--  1 axel  staff    14K May 11 10:34 re.pyc
drwxr-xr-x  2 axel  staff    68B May 11 10:34 site-packages
-rw-r--r--  1 axel  staff    27K May 11 10:34 site.py
-rw-r--r--  1 axel  staff    24K May 11 10:34 site.pyc
lrwxr-xr-x  1 axel  staff    77B May 11 10:34 sre.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/sre.py
lrwxr-xr-x  1 axel  staff    85B May 11 10:34 sre_compile.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/sre_compile.py
-rw-r--r--  1 axel  staff    13K May 11 10:34 sre_compile.pyc
lrwxr-xr-x  1 axel  staff    87B May 11 10:34 sre_constants.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/sre_constants.py
-rw-r--r--  1 axel  staff   6.2K May 11 10:34 sre_constants.pyc
lrwxr-xr-x  1 axel  staff    83B May 11 10:34 sre_parse.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/sre_parse.py
-rw-r--r--  1 axel  staff    20K May 11 10:34 sre_parse.pyc
lrwxr-xr-x  1 axel  staff    78B May 11 10:34 stat.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/stat.py
-rw-r--r--  1 axel  staff   2.9K May 11 10:34 stat.pyc
lrwxr-xr-x  1 axel  staff    79B May 11 10:34 types.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/types.py
-rw-r--r--  1 axel  staff   2.8K May 11 10:34 types.pyc
lrwxr-xr-x  1 axel  staff    82B May 11 10:34 warnings.py -> /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/warnings.py
-rw-r--r--  1 axel  staff    13K May 11 10:34 warnings.pyc
`

axello, 2017-05-23:
I was able to succesfully remove `rm -r ~/.platformio/penv` from my user account, so no hidden root refs.

ivankravets, 2017-06-01:
Please contact me@ikravets.com I will provide further help. Thanks!

ivankravets, 2017-06-03:
Could you try 

```
rm -r ~/.platformio/penv

sudo python -c "$(curl -fsSL https://bootstrap.pypa.io/get-pip.py)"
```

Start Atom

closed 2017-06-27 by ivankravets