platformio/platformio-atom-ide #460: InstallationManager

TranielDan, 2017-05-10:
PIP: /Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:315: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#snimissingwarning.
  SNIMissingWarning

/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:120: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

Exception:
Traceback (most recent call last):
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/basecommand.py", line 209, in main
    status = self.run(options, args)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/commands/install.py", line 310, in run
    wb.build(autobuilding=True)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/wheel.py", line 747, in build
    self.requirement_set.prepare_files(self.finder)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 359, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 576, in _prepare_file
    session=self.session, hashes=hashes)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 809, in unpack_url
    hashes=hashes
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 648, in unpack_http_url
    hashes)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 841, in _download_http_url
    stream=True,
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 480, in get
    return self.request('GET', url, **kwargs)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 377, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 468, in request
    resp = self.send(prep, **send_kwargs)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 576, in send
    r = adapter.send(request, **kwargs)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 46, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 447, in send
    raise SSLError(e, request=request)
SSLError: [Errno 1] _ssl.c:499: error:0D0890A1:asn1 encoding routines:ASN1_verify:unknown message digest algorithm

/Users/Mine/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:120: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

ivankravets, 2017-05-16:
Please open system terminal and provide an output of these commands
```
python --version
pip --version
pip list
echo $PATH
```

Thanks!

ivankravets, 2017-06-01:
Please open PIO IDE Terminal and run
```
pip install -U requests
```

closed 2017-06-01 by ivankravets