platformio/platformio-atom-ide #463: InstallationManager

balucio, 2017-05-12:
Virtualenv Download: Traceback (most recent call last):

  File "/var/folders/0b/bbwr_gh53w3cr_b49ztdlx740000gn/T/tmp-2505wYxTOj3Xwduj/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()

  File "/var/folders/0b/bbwr_gh53w3cr_b49ztdlx740000gn/T/tmp-2505wYxTOj3Xwduj/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)

  File "/var/folders/0b/bbwr_gh53w3cr_b49ztdlx740000gn/T/tmp-2505wYxTOj3Xwduj/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))

  File "/Users/bsaul/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):

  File "/Users/bsaul/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs

    return s.startswith('/')

AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-05-12:
Please open system terminal (`cmd`) and provide:
```
python --version
pip --version
virtualenv -- version
echo $PATH
```

Thanks!

balucio, 2017-05-19:
Hi,

System : MacBoolPro 2009
OS : El Chaptain 10.11.6

 output of command as you required:
$python --version
Python 2.7.10

$ pip --version
-bash: pip: command not found

$ virtualenv -- version
-bash: virtualenv: command not found

$ echo $PATH
/opt/pear/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/opt/ImageMagick/bin:/Applications/Wireshark.app/Contents/MacOS

ivankravets, 2017-05-19:
Could you contact me@ikravets.com and I'll ask you to provide more detailed info. Thanks.

ivankravets, 2017-05-19:
Please close Atom and open system terminal and run
```
rm -rf /Users/bsaul/.platformio/penv
```

Start Atom. Does it work?

closed 2017-06-01 by ivankravets