platformio/platformio-atom-ide #465: Library command failed: lib --global builtin --json-output

SteelPh0enix, 2017-05-14:
Error: Error: spawn platformio ENOENT
    at utils.runPIOCommand.spawnOptions (/home/phoenix/.atom/packages/platformio-ide/lib/library/helpers.js:39:25)
    at onExit (/home/phoenix/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at /home/phoenix/.atom/packages/platformio-ide/lib/utils.js:107:5
    at Function.module.exports.Emitter.simpleDispatch (/usr/lib/atom/node_modules/event-kit/lib/emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (/usr/lib/atom/node_modules/event-kit/lib/emitter.js:141:28)
    at BufferedProcess.handleError (/usr/lib/atom/src/buffered-process.js:369:20)
    at ChildProcess.<anonymous> (/usr/lib/atom/src/buffered-process.js:357:16)
    at emitOne (events.js:96:13)
    at ChildProcess.emit (events.js:188:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:213:12)
    at onErrorNT (internal/child_process.js:359:16)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)

This happens on fresh platformio install (with fresh Atom) under Arch Linux (also updated)

ivankravets, 2017-05-16:
Please open PIO IDE Terminal and type 
```
pio lib --global builtin --json-output
```

Do you see any errors?

ivankravets, 2017-06-01:
http://docs.platformio.org/en/latest/faq.html#platformio-is-not-recognized-as-an-internal-or-external-command

closed 2017-06-01 by ivankravets