platformio/platformio-atom-ide #466: InstallationManager

sones857, 2017-05-14:
Virtualenv Download: Traceback (most recent call last):
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment

    download=download,
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv.py", line 843, in install_wheel

    cert_data = pkgutil.get_data("pip._vendor.requests", "cacert.pem")
  File "/usr/lib/python2.7/pkgutil.py", line 578, in get_data

    loader = get_loader(package)
  File "/usr/lib/python2.7/pkgutil.py", line 464, in get_loader

    return find_loader(fullname)
  File "/usr/lib/python2.7/pkgutil.py", line 474, in find_loader

    for importer in iter_importers(fullname):
  File "/usr/lib/python2.7/pkgutil.py", line 430, in iter_importers

    __import__(pkg)
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/__init__.py", line 13, in <module>

  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/utils/__init__.py", line 26, in <module>
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3130, in <module>
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3116, in _call_aside
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3143, in _initialize_master_working_set

  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 633, in _build_master
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 626, in __init__
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 682, in add_entry
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2120, in find_on_path

  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2510, in from_location
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2827, in _reload_version
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2475, in _version_from_file

  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2643, in _get_metadata
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2011, in get_metadata_lines
  File "/tmp/tmp-9175jgupX6SYc2P3/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2006, in get_metadata
  File "/usr/lib/python2.7/codecs.py", line 296, in decode

    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xb6 in position 147: invalid start byte

ivankravets, 2017-05-16:
Please open system terminal and provide:
```
python --version
pip --version
virtualenv -- version
echo $PATH
```

Thanks!

ivankravets, 2017-06-01:
Please uninstall/install the latest Beta 4

closed 2017-06-01 by ivankravets

sones857, 2017-06-06:
Sorry it took me so long to get back to you, I have installed the latest version as you suggested and everything works perfectly, thank you very much.
Tom