platformio/platformio-atom-ide #467: InstallationManager

Kennyl, 2017-05-16:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/9t/d1wsq_6j0vgbzr1ndxbd_p_h0000gn/T/tmp-1525z9F8zMC9PE6X/virtualenv-15.1.0/virtualenv.py", line 2328, in <module>

    main()

  File "/var/folders/9t/d1wsq_6j0vgbzr1ndxbd_p_h0000gn/T/tmp-1525z9F8zMC9PE6X/virtualenv-15.1.0/virtualenv.py", line 713, in main

    symlink=options.symlink)
  File "/var/folders/9t/d1wsq_6j0vgbzr1ndxbd_p_h0000gn/T/tmp-1525z9F8zMC9PE6X/virtualenv-15.1.0/virtualenv.py", line 925, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/apple/.platformio/penv/bin/../lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/Users/apple/.platformio/penv/bin/../lib/python2.7/posixpath.py", line 54, in isabs

    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-05-16:
> virtualenv-15.1.0/virtualenv.py

It looks that you don't install the latest version of PlatformIO IDE 2.0. Could you navigate to Atom > Settings > Install > Search for `platformio-ide` and install the latest Beta

ivankravets, 2017-05-16:
Please open system terminal and provide:
```
python --version
pip --version
virtualenv -- version
echo $PATH
```

Thanks!

orgkhnargh, 2017-05-19:
Hi @Kennyl,

Can you please let us know if removing the `~/.platformio/penv` directory resolves this issue?

Thank you.

Kennyl, 2017-05-22:
Sorry for last reply
```
Last login: Mon May 22 13:39:30 on ttys000
🍎:💻 [apple]13:40:40 ~
$python --version
Python 2.7.13
🍎:💻 [apple]13:40:44 ~
$pip --version

pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)
🍎:💻 [apple]13:40:56 ~
$
🍎:💻 [apple]13:40:56 ~
$virtualenv -- version
New python executable in version/bin/python
Please make sure you remove any previous custom paths from your /Users/apple/.pydistutils.cfg file.
Installing setuptools, pip...
  Complete output from command /Users/apple/version/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip:
  Ignoring indexes: https://pypi.python.org/simple/
Downloading/unpacking setuptools
Downloading/unpacking pip
Installing collected packages: setuptools, pip
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/virtualenv_support/pip-1.5.5-py2.py3-none-any.whl/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/Library/Python/2.7/site-packages/virtualenv_support/pip-1.5.5-py2.py3-none-any.whl/pip/commands/install.py", line 283, in run
    requirement_set.install(install_options, global_options, root=options.root_path)
  File "/Library/Python/2.7/site-packages/virtualenv_support/pip-1.5.5-py2.py3-none-any.whl/pip/req.py", line 1435, in install
    requirement.install(install_options, global_options, *args, **kwargs)
  File "/Library/Python/2.7/site-packages/virtualenv_support/pip-1.5.5-py2.py3-none-any.whl/pip/req.py", line 671, in install
    self.move_wheel_files(self.source_dir, root=root)
  File "/Library/Python/2.7/site-packages/virtualenv_support/pip-1.5.5-py2.py3-none-any.whl/pip/req.py", line 901, in move_wheel_files
    pycompile=self.pycompile,
  File "/Library/Python/2.7/site-packages/virtualenv_support/pip-1.5.5-py2.py3-none-any.whl/pip/wheel.py", line 215, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/Library/Python/2.7/site-packages/virtualenv_support/pip-1.5.5-py2.py3-none-any.whl/pip/wheel.py", line 209, in clobber
    shutil.copy2(srcfile, destfile)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 130, in copy2
    copyfile(src, dst)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/lib/python2.7/site-packages/easy_install.py'

Storing debug log for failure in /Users/apple/.pip/pip.log
----------------------------------------
...Installing setuptools, pip...done.
Traceback (most recent call last):
  File "/usr/local/bin/virtualenv", line 8, in <module>
    load_entry_point('virtualenv==1.11.5', 'console_scripts', 'virtualenv')()
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 824, in main
    symlink=options.symlink)
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 992, in create_environment
    install_wheel(to_install, py_executable, search_dirs)
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 960, in install_wheel
    'PIP_NO_INDEX': '1'
  File "/Library/Python/2.7/site-packages/virtualenv.py", line 902, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /Users/apple/version/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip failed with error code 2
🍎:💻 [apple]13:41:05 ~
$echo $PATH
``` @ @

orgkhnargh, 2017-05-22:
Oh, the `virtualenv -- version` command isn't right. It should be as follows: `virtualenv --version` (no space after `--`). A virtualenv has been accidentally created under `~/version`. You may want to delete it as you don't actually need it.

Regarding the `IOError: [Errno 13] Permission denied: '/lib/python2.7/site-packages/easy_install.py'`, did you install python via homerew? Please take a look at this issue: Homebrew/legacy-homebrew/issues/44836. They say that removing `/Library/Frameworks/Python.framework` and `~/.pydistutils.cfg` helps. Can you make a backup, remove those files, and install PlatformIO IDE again?

orgkhnargh, 2017-05-22:
@Kennyl I was told that it's not a good idea to remove `/Library/Frameworks/Python.framework`. Please don't follow the instructions from my previous comment.

ivankravets, 2017-06-01:
Try
```
sudo pip install -U setuptools pip
```

ivankravets, 2017-06-03:
Could you try 

```
rm -r ~/.platformio/penv

sudo python -c "$(curl -fsSL https://bootstrap.pypa.io/get-pip.py)"
```

Start Atom

closed 2017-06-27 by ivankravets