platformio/platformio-atom-ide #469: InstallationManager

davestroud, 2017-05-16:
PIP: Could not import setuptools which is required to install from a source distribution.
Traceback (most recent call last):
  File "/Users/davidstroud/.platformio/penv/lib/python2.7/site-packages/pip/req/req_install.py", line 387, in setup_py
    import setuptools  # noqa
  File "/Users/davidstroud/.local/lib/python2.7/site-packages/setuptools/__init__.py", line 10, in <module>
    from six.moves import filter, map
ImportError: No module named six.moves

ivankravets, 2017-05-18:
What is your MacOS version?

davestroud, 2017-05-18:
macOS Sierra
Version 10.12.6 Beta(16G8c)
Processor 2.2 GHz Intel Core i7

ivankravets, 2017-06-01:
Have you resolved this issue?

davestroud, 2017-06-02:
I have! Thank you!!

ivankravets, 2017-06-02:
Could you explain how?

ivankravets, 2017-06-03:
@davestroud Please give us a solution how did you resolve this issue. Thanks!

closed 2017-06-03 by ivankravets