platformio/platformio-atom-ide #475: InstallationManager

baileysage, 2017-05-17:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-29966PKDmmWb5qPE\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-29966PKDmmWb5qPE\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-29966PKDmmWb5qPE\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-29966PKDmmWb5qPE\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-29966PKDmmWb5qPE\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\steinfba\.p...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-05-18:
Please open system terminal (`cmd`) and provide:
```
python --version
pip --version
virtualenv -- version
echo %PATH%
```

Thanks!

baileysage, 2017-05-18:
Python 2.7.13
pip 9.0.1 (ugh, I see this is from the python 3.6 folder. I'll work on that)
echo %PATH% doesn't return anything 
virtualenv gives a "not recognized as the name of...." error

baileysage, 2017-05-18:
pip is still 9.0.1 from the python 2.7 folder

ivankravets, 2017-05-18:
Could you provide an output of 
```
C:/Users/steinfba/.platformio/penv/Scripts/pip
```

Does it use Python 2.7?

baileysage, 2017-05-18:
PS C:\Users\steinfba\.platformio\project-examples\espressif\esp32-espidf-hello-world> C:/Users/steinfba/.platformio/penv/Scripts/pip
The term 'C:/Users/steinfba/.platformio/penv/Scripts/pip' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was includ
ed, verify that the path is correct and try again.
At line:1 char:47
+ C:/Users/steinfba/.platformio/penv/Scripts/pip <<<< 
    + CategoryInfo          : ObjectNotFound: (C:/Users/steinf...env/Scripts/pip:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

baileysage, 2017-05-18:
There is no pip script in the platformio folder.

k4roshi, 2017-05-19:
I solved it by removing .pydistutils.cfg from my home folder

baileysage, 2017-05-19:
@k4roshi Don't seem to have that config file on my system anywhere either. I even uninstalled Atom and both Python versions and reinstalled from scratch. Still getting the same issue. On an older version of PlatformIO on my other Windows machine and have no problem. 

No issues on my Mac so far, but I'll remember your fix if I see this there.

ivankravets, 2017-05-19:
@baileysage could you mail me@ikravets.com? I'll explain further steps. Thanks!

ArunmozhiRajavel, 2017-05-25:
The exact same issue.. @ivankravets  any luck in finding a solution or a workaround??

ivankravets, 2017-05-25:
Guys, please mail me@ikravets.com I'll explain how we can debug this issue.

ivankravets, 2017-06-01:
Try to open system Terminal 
```
pip install virtualenv
```

Restart Atom

baileysage, 2017-06-01:
virtualenv installed correctly. Opened Atom as Administrator, access
denied?

Virtualenv Download: Traceback (most recent call last):
  File
"C:\Users\steinfba\AppData\Local\Temp\tmp-17908KlcF2vuuQQaE\virtualenv-14.0.6\virtualenv.py",
line 2304, in <module>
    main()
  File
"C:\Users\steinfba\AppData\Local\Temp\tmp-17908KlcF2vuuQQaE\virtualenv-14.0.6\virtualenv.py",
line 705, in main
    symlink=options.symlink)
  File
"C:\Users\steinfba\AppData\Local\Temp\tmp-17908KlcF2vuuQQaE\virtualenv-14.0.6\virtualenv.py",
line 924, in create_environment
    download=download,
  File
"C:\Users\steinfba\AppData\Local\Temp\tmp-17908KlcF2vuuQQaE\virtualenv-14.0.6\virtualenv.py",
line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File
"C:\Users\steinfba\AppData\Local\Temp\tmp-17908KlcF2vuuQQaE\virtualenv-14.0.6\virtualenv.py",
line 743, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

ivankravets, 2017-06-01:
1. Close Atom
2. Remove `C:/Users/steinfba/.platformio` directory
3. Open system `cmd` and run `pip install -U pip setuptools virtualenv`
4. Open atom WITHOUT ADMIN RIGHTS

baileysage, 2017-06-01:
Same error. :/

ivankravets, 2017-06-02:
@baileysage could you contact me@ikravets.com ? Thanks!

ivankravets, 2017-06-03:
Please try to disable antivirus and start again

ivankravets, 2017-06-27:
Please reopen if you still need help

baileysage, 2017-06-28:
I figured out that my company restricted exe file from launching from the
temp directory, as that python subprocess was trying to do. I gave him the
"I'm a Dev, give me full access" line and all is well. Sorry to be a butt.

closed 2017-06-28 by ivankravets

ivankravets, 2017-06-28:
1. Please open Atom > Settings > Packages > PlatformIO IDE
2. Check "Advanced > Use development version of PlatformIO IDE"
3. Wait while Atom will re-install PIO IDE

Does it work now?

baileysage, 2017-06-28:
Yes it's working.

ivankravets, 2017-06-28:
Thanks a lot that confirmed! 

Happy coding with PlatformIO!

commit ac7dbefe (2017-06-28): Don't use system temporary directory for virtualenv tool // Resolve #644 Resolve #639 Resolve #630 Resolve #560 Resolve #498 Resolve #492 Resolve #475 Resolve #298