platformio/platformio-atom-ide #477: InstallationManager

jasonoviedo, 2017-05-18:
Conda Virtualenv: An unexpected error has occurred.
Please consider posting the following information to the
conda GitHub issue tracker at:

    https://github.com/conda/conda/issues

Current conda install:

               platform : osx-64
          conda version : 4.3.14
       conda is private : False
      conda-env version : 4.3.14
    conda-build version : not installed
         python version : 3.6.0.final.0
       requests version : 2.12.4
       root environment : /Users/jasonoviedo/anaconda3  (writable)
    default environment : /Users/jasonoviedo/anaconda3
       envs directories : /Users/jasonoviedo/anaconda3/envs
                          /Users/jasonoviedo/.conda/envs
          package cache : /Users/jasonoviedo/anaconda3/pkgs
                          /Users/jasonoviedo/.conda/pkgs
           channel URLs : https://repo.continuum.io/pkgs/free/osx-64
                          https://repo.continuum.io/pkgs/free/noarch
                          https://repo.continuum.io/pkgs/r/osx-64
                          https://repo.continuum.io/pkgs/r/noarch
                          https://repo.continuum.io/pkgs/pro/osx-64
                          https://repo.continuum.io/pkgs/pro/noarch
            config file : None
           offline mode : False
             user-agent : conda/4.3.14 requests/2.12.4 CPython/3.6.0 Darwin/16.5.0 OSX/10.12.4
                UID:GID : 501:20

`$ /Users/jasonoviedo/anaconda3/bin/conda create --yes --quiet python=2 --prefix /Users/jasonoviedo/.platformio/penv`

    Traceback (most recent call last):
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/exceptions.py", line 573, in conda_exception_handler
        return_value = func(*args, **kwargs)
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/cli/main.py", line 134, in _main
        exit_code = args.func(args, p)
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/cli/main_create.py", line 68, in execute
        install(args, parser, 'create')
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/cli/install.py", line 359, in install
        execute_actions(actions, index, verbose=not context.quiet)
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/plan.py", line 800, in execute_actions
        execute_instructions(plan, index, verbose)
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/instructions.py", line 247, in execute_instructions
        cmd(state, arg)
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/instructions.py", line 108, in UNLINKLINKTRANSACTION_CMD
        txn.execute()
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/core/link.py", line 262, in execute
        self.verify()
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/core/link.py", line 241, in verify
        self.prepare()
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/core/link.py", line 148, in prepare
        for pkg_info in self.packages_info_to_link)
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/core/link.py", line 148, in <genexpr>
        for pkg_info in self.packages_info_to_link)
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/core/link.py", line 54, in determine_link_type
        if hardlink_supported(source_test_file, target_prefix):
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/_vendor/auxlib/decorators.py", line 56, in _memoized_func
        result = func(*args, **kwargs)
      File "/Users/jasonoviedo/anaconda3/lib/python3.6/site-packages/conda/gateways/disk/test.py", line 83, in hardlink_supported
        assert isdir(dest_dir), dest_dir
    AssertionError: /Users/jasonoviedo/.platformio/penv

ivankravets, 2017-05-18:
1. Please remove `/Users/jasonoviedo/.platformio/penv`
2. Open system terminal and run `/Users/jasonoviedo/anaconda3/bin/conda create --yes --quiet python=2 --prefix /Users/jasonoviedo/.platformio/penv`

ivankravets, 2017-06-05:
Please try this https://github.com/platformio/platformio-atom-ide/issues/520#issuecomment-306025310

closed 2017-06-05 by ivankravets