platformio/platformio-atom-ide #478: Can't compile file

chaniwat, 2017-05-18:
[image]

ivankravets, 2017-05-18:
Try to remove `~/.platformio` folder and restart build.

chaniwat, 2017-05-18:
I reinstall everything (atom, platformio-ide, python) but it not solved.
run on windows 10

in platformio.ini
[image]

project structue
[image]

chaniwat, 2017-05-18:
Update, try to run outside atom

Can run on command line
[image]

But can't run on powershell
[image]

ivankravets, 2017-05-18:
Please provide an output of `echo %PATH`

chaniwat, 2017-05-19:
```
C:\Python27\;C:\Python27\Scripts;C:\Program Files\Docker\Docker\Resources\bin;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files(x86)\Ampps\php;C:\ProgramData\ComposerSetup\bin;C:\Program Files\Git\cmd;C:\ansicon;C:\Program Files (x86)\Microsoft Emulator Manager\1.0\;C:\Users\Meranote\AppData\Local\Android\sdk\platform-tools;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Users\Meranote\AppData\Roaming\Composer\vendor\bin;C:\Users\Meranote\AppData\Local\atom\bin;C:\Program Files\nodejs\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files\dotnet\;C:\Program Files\Java\jre1.8.0_102\bin;C:\Program Files\Java\jdk1.8.0_102\bin;C:\apache-maven-3.3.9\bin;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\PuTTY\;C:\Users\Meranote\AppData\Roaming\npm;C:\Users\Meranote\AppData\Local\Google\Cloud SDK\google-cloud-sdk\bin;C:\Program Files (x86)\Ampps\mongodb\bin;C:\chromedriver_win32;C:\Program Files\LLVM\bin;C:\Users\Meranote\AppData\Local\atom\bin
```

chaniwat, 2017-07-02:
I found the problem that cause error is "ansicon", i uninstall it and its work now.
(compile/upload via IDE and command line (powershell))

[image]

I installed ansicon as autorun, so everytime i open cmd (or powershell) it hook into it. I think the problem is ansicon, not compatible with powershell (because everytime i try to run pio command via powershell it cause an error but on cmd it work normally). So i uninstall it and it work. 😄

ivankravets, 2017-11-06:
Thanks that reported your solution!

closed 2017-11-06 by ivankravets