platformio/platformio-atom-ide #481: Can not fetch embedded boards

ubIQio, 2017-05-19:
Error: Error: no such option: -f

    at utils.runPIOCommand.cacheValid (/Users/jonathanpeace/.atom/packages/platformio-ide/lib/platform/helpers.js:20:25)
    at /Users/jonathanpeace/.atom/packages/platformio-ide/lib/utils.js:89:46
    at Array.map (native)
    at onExit (/Users/jonathanpeace/.atom/packages/platformio-ide/lib/utils.js:89:36)
    at exit (/Users/jonathanpeace/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:338:1)
    at /Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:351:1
    at Socket.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:220:1)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

ivankravets, 2017-05-20:
Please open system terminal and remove old PIO Core. It looks that you have multiple PIO Cores in a system. Just run in OS Terminal `pip uninstall platformio` or `brew uninstall platformio`. I don't know which way was used to install PIO Core before.

ubIQio, 2017-05-22:
Thanks, Ivan

`pip uninstall platformio`

fixed it

closed 2017-05-22 by ubIQio