platformio/platformio-atom-ide #484: Uncaught Error: Failed to get boards

saucompeng, 2017-05-21:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.17.0 x64
**Electron**: 1.3.15
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.3.3

### Stack Trace

Uncaught Error: Failed to get boards

```
At C:\Users\saucompeng\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactErrorUtils.js:51

Error: Failed to get boards
    at getBoards (/packages/platformio-ide/lib/utils.js:279:13)
    at InitializeNewProjectView.initializeBoardsSelect (/packages/platformio-ide/lib/init/view.js:50:20)
    at InitializeNewProjectView.initialize (/packages/platformio-ide/lib/init/view.js:46:10)
    at InitializeNewProjectView.BaseView (/packages/platformio-ide/lib/base-view.js:16:20)
    at /packages/platformio-ide/lib/init/view.js:16:38)
    at Object.command (/packages/platformio-ide/lib/init/command.js:20:14)
    at /packages/platformio-ide/lib/main.js:100:59)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.17.0/resources/app/src/command-registry.js:265:1)
    at CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.17.0/resources/app/src/command-registry.js:3:1)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (~/AppData/Local/atom/app-1.17.0/resources/app/src/command-registry.js:166:1)
    at Object.runAtomCommand (/packages/platformio-ide/lib/utils.js:252:24)
    at callback (/packages/platformio-ide/lib/home/containers/welcome-page.js:26:31)
    at Object.invokeGuardedCallback (/packages/platformio-ide/node_modules/react-dom/lib/ReactErrorUtils.js:26:5)
    at executeDispatch (/packages/platformio-ide/node_modules/react-dom/lib/EventPluginUtils.js:85:21)
    at Object.executeDispatchesInOrder (/packages/platformio-ide/node_modules/react-dom/lib/EventPluginUtils.js:108:5)
    at executeDispatchesAndRelease (/packages/platformio-ide/node_modules/react-dom/lib/EventPluginHub.js:43:22)
    at executeDispatchesAndReleaseTopLevel (/packages/platformio-ide/node_modules/react-dom/lib/EventPluginHub.js:54:10)
    at Array.forEach (native)
    at forEachAccumulated (/packages/platformio-ide/node_modules/react-dom/lib/forEachAccumulated.js:24:9)
    at Object.processEventQueue (/packages/platformio-ide/node_modules/react-dom/lib/EventPluginHub.js:257:7)
    at runEventQueueInBatch (/packages/platformio-ide/node_modules/react-dom/lib/ReactEventEmitterMixin.js:17:18)
    at /packages/platformio-ide/node_modules/react-dom/lib/ReactEventEmitterMixin.js:28:5)
    at handleTopLevelImpl (/packages/platformio-ide/node_modules/react-dom/lib/ReactEventListener.js:72:24)
    at ReactDefaultBatchingStrategyTransaction.perform (/packages/platformio-ide/node_modules/react-dom/lib/Transaction.js:140:20)
    at Object.batchedUpdates (/packages/platformio-ide/node_modules/react-dom/lib/ReactDefaultBatchingStrategy.js:62:26)
    at Object.batchedUpdates (/packages/platformio-ide/node_modules/react-dom/lib/ReactUpdates.js:97:27)
    at dispatchEvent (/packages/platformio-ide/node_modules/react-dom/lib/ReactEventListener.js:147:20)
```

### Commands

```
  2x -0:12.7.0 platformio-ide:initialize-new-project (atom-pane.pane.active)
```

### Non-Core Packages

```
autocomplete-clang 0.11.3 
build 0.67.0 
busy 0.7.0 
busy-signal 1.4.1 
file-icons 2.1.6 
intentions 1.1.2 
language-ini 1.19.0 
linter 2.1.4 
linter-gcc 0.7.1 
linter-ui-default 1.5.4 
minimap 4.28.2 
platformio-ide 2.0.0-beta.3.3 
platformio-ide-debugger 1.2.0 
platformio-ide-terminal 2.5.1 
tool-bar 1.1.0 
```

Tomibubi, 2017-05-30:
The same to me. With Windows 10.

ivankravets, 2017-05-30:
Do you have multiple PIO Cores in a system?

* http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

Tomibubi, 2017-05-30:
I have only one PIO core. Now a've tried to reinstall Python, but it's the
same.
Tomas

ivankravets, 2017-05-30:
@Tomibubi could you mail me@ikravets.com? I'll send you further instructions. Thanks!

ivankravets, 2017-06-01:
Please open PIO IDE Terminal and type
```
pio boards
```

What do you see?

Tomibubi, 2017-06-02:
See attached file with output of pio boards

But it is after reinstallation if Atom and Platfromio.

�� 
  
 P l a t f o r m :   a t m e l a v r  
  
  
  
 I D                                         M C U                         F r e q u e n c y     F l a s h       R A M         N a m e  
  
  
 b l u e f r u i t m i c r o                 A T M E G A 3 2 U 4           8 M h z             2 8 k B         2 . 5 k B     A d a f r u i t   B l u e f r u i t   M i c r o  
 f e a t h e r 3 2 u 4                       A T M E G A 3 2 U 4           8 M h z             2 8 k B         2 . 5 k B     A d a f r u i t   F e a t h e r  
 f l o r a 8                                 A T M E G A 3 2 U 4           8 M h z             2 8 k B         2 . 5 k B     A d a f r u i t   F l o r a  
 g e m m a                                   A T T I N Y 8 5               8 M h z             8 k B           5 1 2 B       A d a f r u i t   G e m m a  
 m e t r o                                   A T M E G A 3 2 8 P           1 6 M h z           3 1 k B         2 k B         A d a f r u i t   M e t r o  
 p r o t r i n k e t 3 f t d i               A T M E G A 3 2 8 P           1 2 M h z           2 8 k B         2 k B         A d a f r u i t   P r o   T r i n k e t   3 V / 1 2 M H z   ( F T D I )  
 p r o t r i n k e t 3                       A T M E G A 3 2 8 P           1 2 M h z           2 8 k B         2 k B         A d a f r u i t   P r o   T r i n k e t   3 V / 1 2 M H z   ( U S B )  
 p r o t r i n k e t 5 f t d i               A T M E G A 3 2 8 P           1 6 M h z           2 8 k B         2 k B         A d a f r u i t   P r o   T r i n k e t   5 V / 1 6 M H z   ( F T D I )  
 p r o t r i n k e t 5                       A T M E G A 3 2 8 P           1 6 M h z           2 8 k B         2 k B         A d a f r u i t   P r o   T r i n k e t   5 V / 1 6 M H z   ( U S B )  
 t r i n k e t 3                             A T T I N Y 8 5               8 M h z             8 k B           5 1 2 B       A d a f r u i t   T r i n k e t   3 V / 8 M H z  
 t r i n k e t 5                             A T T I N Y 8 5               1 6 M h z           8 k B           5 1 2 B       A d a f r u i t   T r i n k e t   5 V / 1 6 M H z  
 a l t a i r                                 A T M E G A 2 5 6 R F R 2     1 6 M h z           2 4 8 k B       3 2 k B       A l t a i r  
 a r d u b o y                               A T M E G A 3 2 U 4           1 6 M h z           2 8 k B         2 . 5 k B     A r d u b o y  
 a r d u b o y _ d e v k i t                 A T M E G A 3 2 U 4           1 6 M h z           2 8 k B         2 . 5 k B     A r d u b o y   D e v K i t  
 b t a t m e g a 1 6 8                       A T M E G A 1 6 8             1 6 M h z           1 4 k B         1 k B         A r d u i n o   B T   A T m e g a 1 6 8  
 b t a t m e g a 3 2 8                       A T M E G A 3 2 8 P           1 6 M h z           2 8 k B         2 k B         A r d u i n o   B T   A T m e g a 3 2 8  
 d i e c i m i l a a t m e g a 1 6 8         A T M E G A 1 6 8             1 6 M h z           1 4 k B         1 k B         A r d u i n o   D u e m i l a n o v e   o r   D i e c i m i l a   A T m e g a 1 6 8  
 d i e c i m i l a a t m e g a 3 2 8         A T M E G A 3 2 8 P           1 6 M h z           3 0 k B         2 k B         A r d u i n o   D u e m i l a n o v e   o r   D i e c i m i l a   A T m e g a 3 2 8  
 e s p l o r a                               A T M E G A 3 2 U 4           1 6 M h z           2 8 k B         2 . 5 k B     A r d u i n o   E s p l o r a  
 e t h e r n e t                             A T M E G A 3 2 8 P           1 6 M h z           3 1 k B         2 k B         A r d u i n o   E t h e r n e t  
 f i o                                       A T M E G A 3 2 8 P           8 M h z             3 0 k B         2 k B         A r d u i n o   F i o  
 c h i w a w a                               A T M E G A 3 2 U 4           1 6 M h z           2 8 k B         2 . 5 k B     A r d u i n o   I n d u s t r i a l   1 0 1  
 l e o n a r d o                             A T M E G A 3 2 U 4           1 6 M h z           2 8 k B         2 . 5 k B     A r d u i n o   L e o n a r d o  
 l e o n a r d o e t h                       A T M E G A 3 2 U 4           1 6 M h z           2 8 k B         2 . 5 k B     A r d u i n o   L e o n a r d o   E T H  
 l i l y p a d a t m e g a 1 6 8             A T M E G A 1 6 8             8 M h z             1 4 k B         1 k B         A r d u i n o   L i l y P a d   A T m e g a 1 6 8  
 l i l y p a d a t m e g a 3 2 8             A T M E G A 3 2 8 P           8 M h z             3 0 k B         2 k B         A r d u i n o   L i l y P a d   A T m e g a 3 2 8  
 L i l y P a d U S B                         A T M E G A 3 2 U 4           8 M h z             2 8 k B         2 . 5 k B     A r d u i n o   L i l y P a d   U S B  
 m e g a A D K                               A T M E G A 2 5 6 0           1 6 M h z           2 4 8 k B       8 k B         A r d u i n o   M e g a   A D K  
 m e g a a t m e g a 1 2 8 0                 A T M E G A 1 2 8 0           1 6 M h z           1 2 4 k B       8 k B         A r d u i n o   M e g a   o r   M e g a   2 5 6 0   A T m e g a 1 2 8 0  
[... cut ...]
 n u c l e o _ l 4 7 6 r g                   S T M 3 2 L 4 7 6 R G T 6     8 0 M h z           1 0 2 4 k B     1 2 8 k B     S T   N u c l e o   L 4 7 6 R G  
 d i s c o _ f 0 3 0 r 8                     S T M 3 2 F 0 3 0 R 8 T 6     4 8 M h z           6 4 k B         8 k B         S T   S T M 3 2 F 0 3 0 8 D I S C O V E R Y  
 d i s c o _ f 0 5 1 r 8                     S T M 3 2 F 0 5 1 R 8 T 6     4 8 M h z           6 4 k B         8 k B         S T   S T M 3 2 F 0 D I S C O V E R Y  
 d i s c o _ f 3 0 3 v c                 

ivankravets, 2017-06-02:
Could you make clean uninstall of Atom and PIO Core and install again? Thanks!

* http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

ivankravets, 2017-06-05:
Could you open PIO IDE Terminal and run
```
pio boards --json-output
```

?

sunnythaper, 2017-06-05:
`
Error: Traceback (most recent call last):
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 104, in main
    cli(None, None, None)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/platformio/commands/boards.py", line 29, in cli
    return _print_boards_json(query, installed)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/platformio/commands/boards.py", line 99, in _print_boards_json
    boards = _get_boards(installed)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/platformio/commands/boards.py", line 83, in _get_boards
    boards = PlatformManager().get_installed_boards()
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 156, in get_installed_boards
    for manifest in self.get_installed():
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 267, in get_installed
    manifest = self.load_manifest(pkg_dir)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 231, in load_manifest
    src_manifest_path = self.get_src_manifest_path(pkg_dir)
  File "/Users/sthaper/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 202, in get_src_manifest_path
    for item in os.listdir(pkg_dir):
OSError: [Errno 13] Permission denied: '/Users/sthaper/.platformio/platforms/espressif8266'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================`

ivankravets, 2017-06-05:
Please remove close Atom,  remove `/Users/sthaper/.platformio` and start Atom

sunnythaper, 2017-06-05:
Thank you so much, that worked!!

Tomibubi, 2017-06-05:
Here is output of the

pio boards --json-output

Tomas

��[ { " v e n d o r " :   " 4 D S y s t e m s " ,   " r o m " :   5 2 0 1 9 2 ,   " p l a t f o r m " :   " m i c r o c h i p p i c 3 2 " ,   " u r l " :   " h t t p : / / w w w . 4 d s y s t e m s . c o m . a u / p r o d u c t / P i c a d i l l o _ 3 5 T / " ,   " f c p u " :   8 0 0 0 0 0 0 0 ,   " r a m " :   1 3 1 0 7 2 ,   " f r a m e w o r k s " :   [ " a r d u i n o " ] ,   " d e b u g " :   n u l l ,   " m c u " :   " 3 2 M X 7 9 5 F 5 1 2 L " ,   " i d " :   " p i c a d i l l o _ 3 5 t " ,   " n a m e " :   " 4 D S y s t e m s   P I C a d i l l o   3 5 T " } ,   { " v e n d o r " :   " 9 6 B o a r d s " ,   " r o m " :   5 2 4 2 8 8 ,   " p l a t f o r m " :   " s t s t m 3 2 " ,   " u r l " :   " h t t p s : / / d e v e l o p e r . m b e d . o r g / p l a t f o r m s / S T - B 9 6 B - F 4 4 6 V E / " ,   " f c p u " :   1 6 8 0 0 0 0 0 0 ,   " r a m " :   1 3 1 0 7 2 ,   " f r a m e w o r k s " :   [ " m b e d " ] ,   " d e b u g " :   { " t o o l s " :   { " s t l i n k - v 2 - 1 " :   { " o n b o a r d " :   t r u e } } } ,   " m c u " :   " S T M 3 2 F 4 4 6 V E T 6 " ,   " i d " :   " b 9 6 b _ f 4 4 6 v e " ,   " n a m e " :   " 9 6 B o a r d s   B 9 6 B - F 4 4 6 V E " } ,   { " v e n d o r " :   " N X P " ,   " r o m " :   3 2 7 6 8 ,   " p l a t f o r m " :   " n x p l p c " ,   " u r l " :   " h t t p s : / / d e v e l o p e r . m b e d . o r g / h a n d b o o k / m b e d - N X P - L P C 1 1 U 2 4 " ,   " f c p u " :   4 8 0 0 0 0 0 0 ,   " r a m " :   8 1 9 2 ,   " f r a m e w o r k s " :   [ " m b e d " ] ,   " d e b u g " :   { " t o o l s " :   { " c m s i s - d a p " :   { " o n b o a r d " :   t r u e } } } ,   " m c u " :   " L P C 1 1 U 2 4 " ,   " i d " :   " l p c 1 1 u 2 4 _ 3 0 1 " ,   " n a m e " :   " A R M   m b e d   L P C 1 1 U 2 4   ( + C A N ) " } ,   { " v e n d o r " :   " A d a f r u i t " ,   " r o m " :   2 8 6 7 2 ,   " r a m " :   2 5 6 0 ,   " d e b u g " :   n u l l ,   " m c u " :   " A T M E G A 3 2 U 4 " ,   " i d " :   " b l u e f r u i t m i c r o " ,   " n a m e " :   " A d a f r u i t   B l u e f r u i t   M i c r o " ,   " f r a m e w o r k s " :   [ " a r d u i n o " ] ,   " u r l " :   " h t t p s : / / w w w . a d a f r u i t . c o m / p r o d u c t s / 2 6 6 1 " ,   " p l a t f o r m " :   " a t m e l a v r " ,   " f c p u " :   8 0 0 0 0 0 0 } ,   { " v e n d o r " :   " A d a f r u i t " ,   " r o m " :   2 6 2 1 4 4 ,   " p l a t f o r m " :   " a t m e l s a m " ,   " u r l " :   " h t t p s : / / w w w . a d a f r u i t . c o m / p r o d u c t / 3 0 0 0 " ,   " f c p u " :   4 8 0 0 0 0 0 0 ,   " r a m " :   3 2 7 6 8 ,   " f r a m e w o r k s " :   [ " a r d u i n o " ] ,   " d e b u g " :   { " t o o l s " :   { " c m s i s - d a p " :   { } } } ,   " m c u " :   " S A M D 2 1 G 1 8 A " ,   " i d " :   " a d a f r u i t _ c i r c u i t p l a y g r o u n d _ m 0 " ,   " n a m e " :   " A d a f r u i t   C i r c u i t   P l a y g r o u n d   E x p r e s s " } ,   { " v e n d o r " :   " A d a f r u i t " ,   " r o m " :   1 0 4 4 4 6 4 ,   " p l a t f o r m " :   " e s p r e s s i f 3 2 " ,   " u r l " :   " h t t p s : / / w w w . a d a f r u i t . c o m / p r o d u c t / 3 4 0 5 " ,   " f c p u " :   2 4 0 0 0 0 0 0 0 ,   " r a m " :   2 9 4 9 1 2 ,   " f r a m e w o r k s " :   [ " a r d u i n o " ,   " e s p i d f " ] ,   " d e b u g " :   n u l l ,   " m c u " :   " E S P 3 2 " ,   " i d " :   " f e a t h e r e s p 3 2 " ,   " n a m e " :   " A d a f r u i t   E S P 3 2   F e a t h e r " } ,   { " v e n d o r " :   " A d a f r u i t " ,   " r o m " :   2 8 6 7 2 ,   " r a m " :   2 5 6 0 ,   " d e b u g " :   n u l l ,   " m c u " :   " A T M E G A 3 2 U 4 " ,   " i d " :   " f e a t h e r 3 2 u 4 " ,   " n a m e " :   " A d a f r u i t   F e a t h e r " ,   " f r a m e w o r k s " :   [ " a r d u i n o " ] ,   " u r l " :   " h t t p s : / / l e a r n . a d a f r u i t . c o m / a d a f r u i t - f e a t h e r - 3 2 u 4 - b l u e f r u i t - l e / " ,   " p l a t f o r m " :   " a t m e l a v r " ,   " f c p u " :   8 0 0 0 0 0 0 } ,   { " v e n d o r " :   " A d a f r u i t " ,   " r o m " :   2 6 2 1 4 4 ,   " p l a t f o r m " :   " a t m e l s a m " ,   " u r l " :   " h t t p s : / / w w w . a d a f r u i t . c o m / p r o d u c t / 2 7 7 2 " ,   " f c p u " :   4 8 0 0 0 0 0 0 ,   " r a m " :   3 2 7 6 8 ,   " f r a m e w o r k s " :   [ " a r d u i n o " ] ,   " d e b u g " :   { " t o o l s " :   { " c m s i s - d a p " :   { } } } ,   " m c u " :   " S A M D 2 1 G 1 8 A " ,   " i d " :   " a d a f r u i t _ f e a t h e r _ m 0 _ u s b " ,   " n a m e " :   " A d a f r u i t   F e a t h e r   M 0 " } ,   { " v e n d o r " :   " A d a f r u i t " ,   " r o m " :   2 8 6 7 2 ,   " r a m " :   2 5 6 0 ,   " d e b u g " :   n u l l ,   " m c u " :   " A T M E G A 3 2 U 4 " ,   " i d " :   " f l o r a 8 " ,   " n a m e " :   " A d a f r u i t   F l o r a " ,   " f r a m e w o r k s " :   [ " a r d u i n o " ] ,   " u r l " :   " h t t p : / / w w w . a d a f r u i t . c o m / p r o d u c t / 6 5 9 " ,   " p l a t f o r m " :   " a t m e l a v r " ,   " f c p u " :   8 0 0 0 0 0 0 } ,   { " v e n d o r " :   " A d a f r u i t " ,   " r o m " :   8 1 9 2 ,   " r a m " :   5 1 2 ,   " d e b u g " :   n u l l ,   " m c u " :   " A T T I N Y 8 5 " ,   " i d " :   " g e m m a " ,   " n a m e " :   " A d a f r u i t   G e m m a " ,   " f r a m e w o r k s " :   [ " a r d u i n  [... cut ...]  :   5 2 4 2 8 8 ,   " p l a t f o r m " :   " n x p l p c " ,   " u r l " :   " h t t p s : / / d e v e l o p e r . m b e d . o r g / p l a t f o r m s / E A - L P C 4 0 8 8 - D i s p l a y - M o d u l e / " ,   " f c p u " :   1 2 0 0 0 0 0 0 0 ,   " r a m " :   9 8 3 0 4 ,   " f r a m e w o r k s " :   [ " m b e d " ] ,   " d e b u g " :   { " t o o l s " :   { " c m s i s - d a p " :   { " o n b o a r d " :   t r u e } } } ,   " m c u " :   " L P C 4 0 8 8 " ,   " i d " :   " l p c 4 0 8 8 _

ivankravets, 2017-06-05:
@Tomibubi could you update to PIO IDE Beta 6?

Atom > Settings > Updates

Tomibubi, 2017-06-06:
I've updated it now.

ivankravets, 2017-06-06:
Does it work now?

Tomibubi, 2017-06-06:
It works without any problem.

closed 2017-06-06 by ivankravets

pavel1024, 2017-07-06:
@ivankravets 

 hi I have this problem

C/C++ project index rebuild failed
Error: Traceback (most recent call last):
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 104, in main
    cli(None, None, None)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/commands/init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 39, in __init__
    self._gather_tplvars()
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 152, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 134, in __call__
    value = self.func(*args)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 74, in get_project_build_data
    target=["idedata"])
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 109, in cli
    result = (envname, ep.process())
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 193, in process
    result = self._run()
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 287, in _run
    p = PlatformFactory.newPlatform(self.options['platform'])
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 211, in newPlatform
    name, requirements)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 308, in get_package_dir
    manifest = self.get_package(name, requirements, url)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 278, in get_package
    for manifest in self.get_installed():
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 268, in get_installed
    manifest = self.load_manifest(pkg_dir)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 232, in load_manifest
    src_manifest_path = self.get_src_manifest_path(pkg_dir)
  File "/Users/pavel/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 203, in get_src_manifest_path
    for item in os.listdir(pkg_dir):
OSError: [Errno 13] Permission denied: '/Users/pavel/.platformio/platforms/espressif8266'
============================================================
An unexpected error occurred. Further steps:
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
============================================================