platformio/platformio-atom-ide #485: Platform command failed: platform install linux_arm@1.3.0

rlatkdgur109, 2017-05-22:
Error: Error: Could not find a version that satisfies the requirement '~1.40802.0' for your system 'windows_amd64'

    at file:///C:/Users/SHanest/.atom/packages/platformio-ide/lib/platform/helpers.js:61:25
    at onExit (file:///C:/Users/SHanest/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/SHanest/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\SHanest\AppData\Local\atom\app-1.17.0\resources\app\src\buffered-process.js:338:1)
    at ChildProcess.<anonymous> (C:\Users\SHanest\AppData\Local\atom\app-1.17.0\resources\app\src\buffered-process.js:368:1)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

orgkhnargh, 2017-05-22:
Hi @rlatkdgur109 

This seems to be an issue with the `linux_arm` platform itself, rather than with the IDE. Please consider moving this issue to the platform issues tracker: https://github.com/platformio/platform-linux_arm/issues

ivankravets, 2017-06-01:
Duplicate of https://github.com/platformio/platform-linux_arm/issues/2

closed 2017-06-01 by ivankravets