platformio/platformio-atom-ide #486: InstallationManager

nano8ot, 2017-05-22:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/r1/xf1c9xm55fl7hj28g_qfz1c80000gn/T/tmp-2896kWXyHmgF9ZMy/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/r1/xf1c9xm55fl7hj28g_qfz1c80000gn/T/tmp-2896kWXyHmgF9ZMy/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/r1/xf1c9xm55fl7hj28g_qfz1c80000gn/T/tmp-2896kWXyHmgF9ZMy/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment

    download=download,

  File "/var/folders/r1/xf1c9xm55fl7hj28g_qfz1c80000gn/T/tmp-2896kWXyHmgF9ZMy/virtualenv-14.0.6/virtualenv.py", line 877, in install_wheel

    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "/var/folders/r1/xf1c9xm55fl7hj28g_qfz1c80000gn/T/tmp-2896kWXyHmgF9ZMy/virtualenv-14.0.6/virtualenv.py", line 783, in call_subprocess

    % (cmd_desc, proc.returncode))
OSError: Command /Users/jacek/.platformio/penv/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-06-01:
Please provide an output from system Terminal:
```
echo $PATH
python --version
pip --version
pip list
```

ivankravets, 2017-06-03:
Could you try 

```
rm -r ~/.platformio/penv

sudo python -c "$(curl -fsSL https://bootstrap.pypa.io/get-pip.py)"
```

Start Atom

ivankravets, 2017-06-27:
Please reopen if you still need help

closed 2017-06-27 by ivankravets

me21, 2017-10-27:
It was fixed for me by installing virtualenv package in Python manually.