platformio/platformio-atom-ide #488: InstallationManager

egn-dev, 2017-05-23:
My previous segnalation (for platformio-ide 2.0.0-beta.2) was closed and marked as duplicate and solved.
Now, trying to install platformio-ide 2.0.0-beta.3 I get the same error as before:

Virtualenv Download: Traceback (most recent call last):
  File "C:/Users/admin/.platformio/penv/lib/python2.7/site.py", line 703, in <module>
    main()
  File "C:/Users/admin/.platformio/penv/lib/python2.7/site.py", line 692, in main
    aliasmbcs()
  File "C:/Users/admin/.platformio/penv/lib/python2.7/site.py", line 515, in aliasmbcs
    import locale, codecs
  File "C:\Users\admin\.platformio\penv\lib\python2.7/locale.py", line 17, in <module>
    import functools
ImportError: No module named functools

Following details about my system:
Windows 10 (1607) x64
Python 3.6.1 x64 (C:\Program Files\Python36) - No Python 2.x
Atom 1.16.0 x64

egn-dev, 2017-05-25:
I managed to install platformio-ide only after manually installing Python 2.7.13 in its default directory "C:\Python27" (without adding it to "Path" environment path)

ivankravets, 2017-06-01:
Could you run?
```
C:/msys64/usr/bin/python2.exe -c "import functools; import ConfigParser; import sys; print ".".join(str(v) for v in sys.version_info[:2])"
```

ivankravets, 2017-06-01:
@stefan-schindler do you also have `C:/msys64/usr/bin/python2.exe`?

stefan-schindler, 2017-06-01:
@ivankravets No, but here `C:/msys64/mingw32/bin/` (this is added to PATH when **msys** is instaled) I have theses files:
[image]

stefan-schindler, 2017-06-01:
And when I call the `command` you provided above, it runs ok (it outputs `2.7`). The `functools` module isn't missing from **msys** python interpereter, but from the Python which isn't successfully created using `virtualenv.py` (which is executed by **msys** python interpreter).

closed 2017-06-03 by ivankravets

raxetul, 2017-06-16:
Platformio-ide package under atom seems to want the python.exe under a python directory. Because of this it fails.
[image]

commit 9f74581e (2017-06-03): Ignore MSYS's Python // Resolve #439 Resolve #488
commit bf518816 (2017-06-03): Ignore MSYS's Python // Resolve #439 Resolve #488