platformio/platformio-atom-ide #489: Library command failed: lib --global stats --json-output

ChristopherScot, 2017-05-23:
Internet access seems to be required to do anything and everything with platform.io. I am using the internet at my work and for whatever reason (probably because atom/platform.io aren't picking up the intermediary certificates that are used by chrome and Edge to access the internet here) it won't download any updates, or allow me to start a new project.

Error: Error: You are not connected to the Internet

    at utils.runPIOCommand.spawnOptions (file:///C:/Users/P00111924/.atom/packages/platformio-ide/lib/library/helpers.js:39:25)
    at onExit (file:///C:/Users/P00111924/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/P00111924/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\P00111924\AppData\Local\atom\app-1.17.0\resources\app\src\buffered-process.js:338:1)
    at C:\Users\P00111924\AppData\Local\atom\app-1.17.0\resources\app\src\buffered-process.js:359:1
    at Socket.<anonymous> (C:\Users\P00111924\AppData\Local\atom\app-1.17.0\resources\app\src\buffered-process.js:220:1)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

ivankravets, 2017-06-01:
Open PIO IDE Terminal and type
```
pio settings set enable_ssl false
```

closed 2017-06-01 by ivankravets

ChristopherScot, 2017-06-01:
You are my hero!

Thank you!!

Christopher Martin
(954) 673-1694
Christopher.Scot.Martin@gmail.com
706 East River Dr.
Margate, FL 33401