platformio/platformio-atom-ide #491: InstallationManager

quando1, 2017-05-24:
PIP: System nie mo�e odnale�� okre�lonej �cie�ki.

ivankravets, 2017-05-27:
Please mail me@ikravets.com for the further instructions. Thanks!

ivankravets, 2017-06-01:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

quando1, 2017-06-01:
Thank you for the reply and please help

W dniu 2017-06-01 o 21:07, Ivan Kravets pisze:

>
> Please provide an output from system Terminal |cmd|
>
> |echo %PATH% python --version pip --version pip list |

ivankravets, 2017-06-01:
See comment above. Please provide all information https://github.com/platformio/platformio-atom-ide/issues/491#issuecomment-305590555

quando1, 2017-06-02:
Microsoft Windows [Version 6.3.9600]
(c) 2013 Microsoft Corporation. Wszelkie prawa zastrzeżone.

C:\Windows\System32>echo %PATH%
C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\Program
Files (x86)\HP SimplePass\;C:\Program Files (x86)\AMD APP\bin\x86_64;C:\Program
Files (x86)\AMD APP\bin\x86;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\W
bem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;;C:\Program Files (x86)\ATI Tech
nologies\ATI.ACE\Core-Static;

C:\Windows\System32>python --version
Python 2.7.13

C:\Windows\System32>pip --version
pip 9.0.1 from c:\python27\lib\site-packages (python 2.7)

Microsoft Windows [Version 6.3.9600]
(c) 2013 Microsoft Corporation. Wszelkie prawa zastrzeżone.

C:\Windows\System32>pip list
DEPRECATION: The default format will switch to columns in the future. You can us
e --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.con
f under the [list] section) to disable this warning.
appdirs (1.4.3)
packaging (16.8)
pip (9.0.1)
pyparsing (2.2.0)
setuptools (35.0.2)
six (1.10.0)

ivankravets, 2017-06-02:
Could you remove `.platformio` hidden folder from Users Home folder and restart atom?

quando1, 2017-06-02:
Yes I deleted the folder and started Atom and the same pip message: The system can not find the specified path

ivankravets, 2017-06-02:
Could you mail me@ikravets.com for details?

ivankravets, 2017-06-03:
Resolved. The problem was with Comodo antivirus

closed 2017-06-03 by ivankravets