platformio/platformio-atom-ide #492: InstallationManager

comada, 2017-05-24:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-1132BUTIPrhHf4V2\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-1132BUTIPrhHf4V2\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-1132BUTIPrhHf4V2\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-1132BUTIPrhHf4V2\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-1132BUTIPrhHf4V2\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\dep07551\.p...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

comada, 2017-05-24:
Platformio doesn't work, always getting above installation error

ivankravets, 2017-05-27:
Please mail me@ikravets.com for the further instructions. Thanks!

ivankravets, 2017-06-01:
Does it work now?

ivankravets, 2017-06-01:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

comada, 2017-06-01:
I don't have acess to my Arduino board, yet. 
Can check it earliest next Tue. Will inform you!
Thanks for supporting this!

comada, 2017-06-06:
Hi,

I just have updated to platformio IDE 2.0.0.0 beta 6

 

I still get an error, while trying to upload to arduino due:

 

[06/06/17 09:30:08] Processing due (platform: atmelsam; board: due; framework: arduino)

PackageManager: Installing tool-bossac @ ~1.10700.0

Error: Detected unknown package 'tool-bossac'

 

 

This is the asked output:

PS C:\Data\Arduino\EAP-control_01> echo %PATH%

%PATH%

PS C:\Data\Arduino\EAP-control_01> python --version

Python 2.7.13

PS C:\Data\Arduino\EAP-control_01> pip --version

pip 9.0.1 from c:\users\dep07551\.platformio\penv\lib\site-packages (python 2.7)

PS C:\Data\Arduino\EAP-control_01> pip list

 

 

Best

 Achim

 

 

 

esendet: Donnerstag, 01. Juni 2017 um 21:06 Uhr

ivankravets, 2017-06-06:
Please remove `c:\users\dep07551\.platformio` and restart Atom

comada, 2017-06-07:
Hi,

 

I renamen the folder and started Atom.

Results:

 

1) A new .platformio folder has been created

2) I get two error messages while starting Atom:

 

C/C++ project index rebuild failed
'platformio' is not recognized as an internal or external command,
operable program or batch file.

 

InstallationManager
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-9396fBat4XhJ99ZO\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-9396fBat4XhJ99ZO\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-9396fBat4XhJ99ZO\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-9396fBat4XhJ99ZO\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\dep07551\AppData\Local\Temp\tmp-9396fBat4XhJ99ZO\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\dep07551\.p...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2
Report a problem

 

 

3) Neither the built function (which worked before) nor the upload function do work.

 

Best

 

 

 

Gesendet: Dienstag, 06. Juni 2017 um 16:01 Uhr

closed 2017-06-28 by ivankravets

ivankravets, 2017-06-28:
1. Please open Atom > Settings > Packages > PlatformIO IDE
2. Check "Advanced > Use development version of PlatformIO IDE"
3. Wait while Atom will re-install PIO IDE

Does it work now?

commit ac7dbefe (2017-06-28): Don't use system temporary directory for virtualenv tool // Resolve #644 Resolve #639 Resolve #630 Resolve #560 Resolve #498 Resolve #492 Resolve #475 Resolve #298