platformio/platformio-atom-ide #493: Same as #94 but on latest stable and beta Platformio

sdaitzman, 2017-05-24:
Let me know if you have any questions! I'd really appreciate help getting this fixed. Thank you for all of your work so far on this project.

[Enter steps to reproduce:]

1. Try to run Platformio

**Atom**: 1.17.0 x64
**Electron**: 1.3.15
**OS**: Mac OS X 10.12.5
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.3

### Stack Trace

Failed to activate the platformio-ide package

```
At Path must be a string. Received undefined

TypeError: Path must be a string. Received undefined
    at assertPath (path.js:7:11)
    at Object.join (path.js:1213:7)
    at /packages/platformio-ide/lib/config.js:112:84)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:1)
    at /packages/platformio-ide/lib/main.js:25:15)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:73:1)
    at Package.module.exports.Package.requireMainModule (/Applications/Atom.app/Contents/Resources/app/src/package.js:887:1)
    at Package.module.exports.Package.activateNow (/Applications/Atom.app/Contents/Resources/app/src/package.js:242:1)
    at /Applications/Atom.app/Contents/Resources/app/src/package.js:1014:1
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:1)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:1)
    at PackageManager.module.exports.PackageManager.triggerDeferredActivationHooks (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:673:1)
    at /Applications/Atom.app/Contents/Resources/app/src/package-manager.js:604:1
```

### Commands

```
     -1:05.9.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling.tinted-tab-bar.panel-contrast.compact-tree-view.blend-tree-view)
```

### Non-Core Packages

```
activate-power-mode 1.4.0 
aesthetic-ui 3.0.1 
aligner-python 1.1.0 
arduino-upload 1.0.3 
ask-stack 2.2.0 
atom-beautify 0.29.24 
atom-handlebars 1.3.0 
atom-jade 0.3.0 
atom-material-syntax 1.0.6 
atom-material-ui 1.3.10 
atom-soda-dark-ui 1.1.0 
atom-ternjs 0.18.3 
auto-indent 0.5.0 
autocomplete-clang 0.11.3 
autocomplete-python 1.8.63 
browser-plus 0.0.87 
build 0.68.0 
build-gulp 0.11.0 
build-make 0.13.0 
busy 0.7.0 
busy-signal 1.4.1 
cake-syntax 1.0.2 
chester-atom-syntax 0.1.1 
color-picker 2.2.5 
dracula-syntax 2.0.5 
dracula-theme 2.0.3 
duotone-dark-sea-syntax 2.1.0 
duotone-dark-space-syntax 2.1.0 
duotone-dark-syntax 2.1.0 
duotone-light-syntax 2.1.0 
editorconfig 2.2.2 
emmet 2.4.3 
erb-helper 1.1.0 
express-complete 0.2.0 
file-icons 2.1.6 
flatwhite-syntax 0.8.1 
git-control 0.9.0 
git-plus 7.8.0 
git-time-machine 1.5.9 
goto 1.8.3 
graphite-ui 0.15.1 
hey-pane 1.0.0 
highlight-line 0.12.0 
Hydrogen 1.14.1 
hyperclick undefined 
intentions 1.1.2 
intentions-colorpicker 1.0.1 
jonikorpi-atom-solarized-syntax 0.2.4 
jonikorpi-atom-theme-ui 4.4.1 
jquery-snippets 11.0.0 
jshint 1.8.6 
jslint 1.5.1 
language-arduino 0.4.3 
language-docker 1.1.7 
language-fish-shell 1.0.5 
language-ini 1.19.0 
language-julia 0.11.0 
language-pug 0.0.21 
linter 2.1.4 
linter-alex 4.0.0 
linter-gcc 0.7.1 
linter-js-standard 3.9.1 
linter-python-pep8 0.2.0 
linter-ui-default 1.2.4 
markdown-scroll-sync 2.1.2 
merge-conflicts 1.4.5 
minimap 4.28.2 
minimap-bookmarks 0.4.2 
minimap-codeglance 0.4.6 
minimap-color-highlight 4.1.4 
minimap-find-and-replace 4.5.2 
minimap-git-diff 4.3.1 
minimap-pigments 0.2.2 
minimap-selection 4.4.0 
monokai-light 0.4.1 
native-ui 0.20.6 
nord-atom-syntax 0.9.1 
nord-atom-ui 0.11.0 
phix-syntax 0.1.1 
pigments 0.39.1 
plain-theme 0.7.0 
platformio-ide 2.0.0-beta.3 
platformio-ide-terminal 2.5.1 
polymorph-ui 1.8.8 
processing 0.13.0 
processing-language 0.2.0 
python-jedi 0.3.8 
python-tools 0.6.9 
react 0.16.2 
rhino-python 0.8.4 
script 3.15.0 
shadow-syntax 0.1.0 
snippets-jade 0.2.2 
spacegray-dark-ui 0.13.0 
spacegray-light-ui 0.1.0 
standardjs-snippets 2.6.0 
tag 0.5.0 
tool-bar 1.1.0 
tree-view-finder 0.2.1 
turbo-javascript 1.4.0 
unity-dark-ui 2.0.9 
unity-ui 2.1.11 
vim-mode 0.66.0 
```

orgkhnargh, 2017-05-24:
Hi @sdaitzman 

Please open dev tools (Ctrl + Shift + I), navigate to the Console tab and run the following command: `require('fs-plus').getHomeDirectory()`. Let me know what it returns.

sdaitzman, 2017-05-24:
undefined 😢

orgkhnargh, 2017-05-24:
`echo $HOME`?

closed 2017-05-24 by ivankravets

sdaitzman, 2017-05-24:
`echo $HOME` does return my home directory (the absolute filepath: `/Users/` followed by my username)

ivankravets, 2017-05-24:
@sdaitzman Does 2.0.beta.4 works for you?

sdaitzman, 2017-05-24:
@ivankravets @orgkhnargh sadly, it does not; here is the error report from the reporter:

**Atom**: 1.17.0 x64
**Electron**: 1.3.15
**OS**: Mac OS X 10.12.5
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.4

### Stack Trace

Failed to activate the platformio-ide package

```
At Cannot read property 'includes' of undefined

TypeError: Cannot read property 'includes' of undefined
    at Object.updateOSEnviron (/packages/platformio-ide/lib/maintenance.js:42:26)
    at PlatformIOIDEPackage.activate (/packages/platformio-ide/lib/main.js:62:19)
    at Package.module.exports.Package.activateNow (/Applications/Atom.app/Contents/Resources/app/src/package.js:253:1)
    at /Applications/Atom.app/Contents/Resources/app/src/package.js:1014:1
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:1)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:1)
    at /Applications/Atom.app/Contents/Resources/app/src/package-manager.js:655:1
    at Set.forEach (native)
    at PackageManager.module.exports.PackageManager.activatePackage (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:653:1)
    at /Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:519:1
    at exit (/Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:128:1)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:338:1)
    at /Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:351:1
    at Socket.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:220:1)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
```

### Commands

```
     -1:47.2.0 settings-view:open (input.hidden-input)
     -1:42.6.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
activate-power-mode 1.4.0 
aesthetic-ui 3.0.1 
aligner-python 1.1.0 
arduino-upload 1.0.3 
ask-stack 2.2.0 
atom-beautify 0.29.24 
atom-handlebars 1.3.0 
atom-jade 0.3.0 
atom-material-syntax 1.0.6 
atom-material-ui 2.0.4 
atom-soda-dark-ui 1.1.0 
atom-ternjs 0.18.3 
auto-indent 0.5.0 
autocomplete-clang 0.11.3 
autocomplete-python 1.8.63 
browser-plus 0.0.87 
build 0.68.0 
build-gulp 0.11.0 
build-make 0.13.0 
busy 0.7.0 
busy-signal 1.4.1 
cake-syntax 1.0.2 
chester-atom-syntax 0.1.1 
color-picker 2.2.5 
dracula-syntax 2.0.5 
dracula-theme 2.0.3 
duotone-dark-sea-syntax 2.1.0 
duotone-dark-space-syntax 2.1.0 
duotone-dark-syntax 2.1.0 
duotone-light-syntax 2.1.0 
editorconfig 2.2.2 
emmet 2.4.3 
erb-helper 1.1.0 
express-complete 0.2.0 
file-icons 2.1.6 
flatwhite-syntax 0.8.1 
git-control 0.9.0 
git-plus 7.9.2 
git-time-machine 1.5.9 
goto 1.8.3 
graphite-ui 0.15.1 
hey-pane 1.0.0 
highlight-line 0.12.0 
Hydrogen 1.14.1 
hyperclick undefined 
intentions 1.1.2 
intentions-colorpicker 1.0.1 
jonikorpi-atom-solarized-syntax 0.2.4 
jonikorpi-atom-theme-ui 4.4.1 
jquery-snippets 11.0.0 
jshint 1.8.6 
jslint 1.5.1 
language-arduino 0.4.3 
language-docker 1.1.7 
language-fish-shell 1.0.5 
language-ini 1.19.0 
language-julia 0.11.0 
language-pug 0.0.21 
linter 2.1.4 
linter-alex 4.0.0 
linter-gcc 0.7.1 
linter-js-standard 3.9.1 
linter-python-pep8 0.2.0 
linter-ui-default 1.6.0 
markdown-scroll-sync 2.1.2 
merge-conflicts 1.4.5 
minimap 4.28.2 
minimap-bookmarks 0.4.2 
minimap-codeglance 0.4.6 
minimap-color-highlight 4.1.4 
minimap-find-and-replace 4.5.2 
minimap-git-diff 4.3.1 
minimap-pigments 0.2.2 
minimap-selection 4.4.0 
monokai-light 0.4.1 
native-ui 0.20.6 
nord-atom-syntax 0.9.1 
nord-atom-ui 0.11.0 
phix-syntax 0.1.1 
pigments 0.39.1 
plain-theme 0.7.0 
platformio-ide 2.0.0-beta.4 
polymorph-ui 1.8.8 
processing 0.13.0 
processing-language 0.2.0 
python-jedi 0.3.8 
python-tools 0.6.9 
react 0.16.2 
rhino-python 0.8.4 
script 3.15.0 
shadow-syntax 0.1.0 
snippets-jade 0.2.2 
spacegray-dark-ui 0.13.0 
spacegray-light-ui 0.1.0 
standardjs-snippets 2.6.0 
tag 0.5.0 
tool-bar 1.1.0 
tree-view-finder 0.2.1 
turbo-javascript 1.4.0 
unity-dark-ui 2.0.9 
unity-ui 2.1.11 
vim-mode 0.66.0 
```

ivankravets, 2017-05-24:
@sdaitzman Could you configure default shell? Please open system terminal and 
```
chsh -s /bin/bash
```

http://osxdaily.com/2012/03/21/change-shell-mac-os-x/

sdaitzman, 2017-05-25:
@ivankravets changing my shell to bash fixes the problem, but I am a Fish Shell user. I'm assuming some script is getting the home directory that cannot run as intended in fish. If this is the case, it should be run by /bin/bash or /bin/sh explicitly. Could you change this, or could you point me to the source of the error and I can?

ivankravets, 2017-05-25:
@sdaitzman I use Fish too on Mac and don't have any problems.

Have you tried ?
```
brew install fish
chsh -s /usr/local/bin/fish
```

sdaitzman, 2017-05-28:
@ivankravets I tried reinstalling Fish, `chsh`ing it back and no luck. Atom still errors on start. I'm wondering if it's something in my fish config? I have other, similar errors from other packages including the tabs package.

sdaitzman, 2017-05-28:
I commented out Oh-My-Fish in my config.fish and now I have no Atom errors... I suspect that OMF was somehow changing the behavior of whichever shell utility was determining the home directory. I'm creating an issue on the OMF repository and linking it to here: https://github.com/oh-my-fish/oh-my-fish/issues/517

georgik, 2017-06-04:
Same problem on Linux with ZSH. After changing to Bash the PlatformIO works.
Then I've changed default shell back to ZSH and PlatformIO still works. Weird.

commit 5ae53d36 (2017-05-24): Use tilde when can’t determine home directory // Resolve #493
commit cdbf6eec (2018-03-03): Merge branch 'develop'

* develop: (66 commits)
  Prepare 2.0.0-beta.4 release
  Cosmetic changes
  Depend on PlatformIO IDE Debugger >=1.2.0
  Use tilde when can’t determine home directory // Resolve #493
  Update history
  Update license to Apache 2.0
  Bump version to 2.0.0-beta.3.3
  Fix rebuild error when changing project paths (#480)
  Renamed “Featured” column to “Trending” for Library Registry Stats
  Bump version to 2.0.0-beta.3.2
  Refactor C/C++ index rebuild (#474)
  Make sure state is an object (#473)
  Remove debug code
  Bump version to 2.0.0-beta.3.1
  Use goTo from “core”
  Mi [... cut]