platformio/platformio-atom-ide #498: InstallationManager

jmaniac, 2017-05-25:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\jmaniac\AppData\Local\Temp\tmp-13972ARyGEHaZXmu6\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\jmaniac\AppData\Local\Temp\tmp-13972ARyGEHaZXmu6\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\jmaniac\AppData\Local\Temp\tmp-13972ARyGEHaZXmu6\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\jmaniac\AppData\Local\Temp\tmp-13972ARyGEHaZXmu6\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\jmaniac\AppData\Local\Temp\tmp-13972ARyGEHaZXmu6\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\E482158\.pl...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-06-01:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

ronanpaixao, 2017-06-02:
I'm also having trouble with PlatformIO on Windows 10 with an existing Python install (Anaconda with Python 3.6.0). I can't really do anything that requires contacting PIO servers, like even using the debugger, which is quite annoying. I also couldn't register an account without some ugly hacks.

Both when registering or logging in with the IDE I get similar tracebacks:

```
Traceback (most recent call last):
  File "main.py", line 12, in init main (C:\users\vagrant\appdata\local\temp\tmptehjmh\main.c:3307)
  File "C:\Users\Ronan\.platformio\penv\lib\site-packages\platformio\app.py", line 24, in <module>
    import requests
  File "C:\Users\Ronan\.platformio\penv\lib\site-packages\requests\__init__.py", line 48, in <module>
    import urllib3
  File "C:\Users\Ronan\.platformio\penv\lib\site-packages\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "C:\Users\Ronan\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 29, in <module>
    from .connection import (
  File "C:\Users\Ronan\.platformio\penv\lib\site-packages\urllib3\connection.py", line 39, in <module>
    from .util.ssl_ import (
  File "C:\Users\Ronan\.platformio\penv\lib\site-packages\urllib3\util\__init__.py", line 6, in <module>
    from .ssl_ import (
  File "C:\Users\Ronan\.platformio\penv\lib\site-packages\urllib3\util\ssl_.py", line 4, in <module>
    import hmac
  File "C:\Users\Ronan\.platformio\penv\lib\hmac.py", line 8, in <module>
    from operator import _compare_digest as compare_digest
ImportError: cannot import name _compare_digest
```

What I found out is that the IDE is trying to use the system's Python interpreter, in an apparent language clash. In the terminal:

```
PS C:\workspace> python --version
Python 3.6.0 :: Anaconda 4.3.1 (64-bit)
PS C:\workspace> pip --version
pip 9.0.1 from C:\Users\MyUser\.platformio\penv\lib\site-packages (python 2.7)
```

The hack I used to register an account was to do it in the terminal, after properly setting my Path environment variable to look for `C:\Users\MyUser\.platformio\penv` first:

```
PS C:\workspace> $env:Path = "C:\Users\MyUser\.platformio\penv;" + $env:Path
```
and then running `pio account register`.

However, `pio debug` still won't work, probably because the terminal won't talk to the IDE.

ronanpaixao, 2017-06-02:
I found another ugly workaround to get the GUI part working, which consists of opening a terminal, changing the path environment variable as above and then run Atom *from that terminal*.

Still ugly, but gets stuff almost usable.

ivankravets, 2017-06-27:
Have you resolved this issue?

closed 2017-06-28 by ivankravets

ivankravets, 2017-06-28:
1. Please open Atom > Settings > Packages > PlatformIO IDE
2. Check "Advanced > Use development version of PlatformIO IDE"
3. Wait while Atom will re-install PIO IDE

Does it work now?

commit ac7dbefe (2017-06-28): Don't use system temporary directory for virtualenv tool // Resolve #644 Resolve #639 Resolve #630 Resolve #560 Resolve #498 Resolve #492 Resolve #475 Resolve #298