platformio/platformio-atom-ide #499: InstallationManager

quando1, 2017-05-25:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\quand\AppData\Local\Temp\tmp-7844f8Ly8OdgcjDZ\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\quand\AppData\Local\Temp\tmp-7844f8Ly8OdgcjDZ\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\quand\AppData\Local\Temp\tmp-7844f8Ly8OdgcjDZ\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\quand\AppData\Local\Temp\tmp-7844f8Ly8OdgcjDZ\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\quand\AppData\Local\Temp\tmp-7844f8Ly8OdgcjDZ\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\quand\.plat...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-05-27:
Please mail me@ikravets.com for detailed instructions. Thanks!

ivankravets, 2017-06-01:
Please provide an output from `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

ivankravets, 2017-06-27:
Please reopen if you still need help

closed 2017-06-27 by ivankravets

me21, 2017-10-27:
It was fixed for me by installing virtualenv package in Python manually.