platformio/platformio-atom-ide #500: C/C++ project index rebuild failed

mongrol, 2017-05-25:
New dev machine. Updated PIO to latest release after some months. I get the following error when trying to rebuild the Project index. 

sh: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

sh: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

PlatformioException: Could not load broken JSON: /home/mongrol/.platformio/lib/Artnet/library.json:
File "/home/mongrol/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 153:
env.SConscript("$BUILD_SCRIPT")
File "/home/mongrol/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/mongrol/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/home/mongrol/.platformio/platforms/teensy/builder/main.py", line 212:
target_elf = env.BuildProgram()
File "/home/mongrol/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/mongrol/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 65:
deplibs = env.BuildDependentLibraries("$PROJECTSRC_DIR")
File "/home/mongrol/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/mongrol/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 656:
lib_builders = env.GetLibBuilders()
File "/home/mongrol/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/mongrol/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 628:
env, join(libs_dir, item), verbose=verbose)
File "/home/mongrol/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 50:
env, path, verbose=verbose)
File "/home/mongrol/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 95:
self._manifest = manifest if manifest else self.load_manifest()
File "/home/mongrol/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 514:
manifest = util.load_json(join(self.path, "library.json"))
File "/home/mongrol/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 178:
file_path)

========================== [ERROR] Took 0.92 seconds ==========================

orgkhnargh, 2017-05-25:
> PlatformioException: Could not load broken JSON: /home/mongrol/.platformio/lib/Artnet/library.json:

Could you please check the contents of the `/home/mongrol/.platformio/lib/Artnet/library.json` file? Looks like it's not a valid JSON file.

mongrol, 2017-05-27:
Deleted the Artnet package and project reindexed fine. Thanks.

ivankravets, 2017-05-27:
> /lib/Artnet

Where did you find this library?

mongrol, 2017-05-28:
No idea. I didn't manually download it and my project doesn't use it. 
PIO must have at some point in the last 6 months.

closed 2017-06-01 by ivankravets