platformio/platformio-atom-ide #505: InstallationManager

studuino, 2017-05-26:
Virtualenv Download: Traceback (most recent call last):
  File "C:/Users/administrator/.platformio/penv/lib/python2.7/site.py", line 703, in <module>
    main()
  File "C:/Users/administrator/.platformio/penv/lib/python2.7/site.py", line 692, in main
    aliasmbcs()
  File "C:/Users/administrator/.platformio/penv/lib/python2.7/site.py", line 515, in aliasmbcs
    import locale, codecs
  File "C:\Users\administrator\.platformio\penv\lib\python2.7/locale.py", line 19, in <module>
    import functools
ImportError: No module named functools

stefan-schindler, 2017-05-26:
Hello. Do you have MSYS or MinGW installed on your system, please?

studuino, 2017-05-26:
Yes, I have them installed sir.

stefan-schindler, 2017-05-26:
Thanks :) If you want to be updated on this issue you can subscribe to this Pull Request disscussion:  #439

studuino, 2017-05-27:
Hi stefan-schindler,
Thanks for your support.

closed 2017-05-27 by studuino